OrcaSlicer/OrcaSlicer #7017: First layer toolchange uses regular retraction lengths, not toolchange lengths

staal54a, 2024-10-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta2

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Modified Artist-D Pro

### How to reproduce

1. Open a project with a multi-extruder printer that has different toolchange retraction lengths from the normal retraction lengths
2. Add two objects to the build plate assigned to two different extruders. Ensure they are both touching the build plate
3. Slice the file

### Actual results

When I do the above, I notice that when the initial_tool (the one that starts printing first) changes to the next tool, the next tool extrudes its normal retraction length, not the toolchange length.

On every other layer besides the first one, the toolchange retraction length is used correctly. See the following two screenshots for the problematic first layer and a normal next layer. Note that the normal retraction for T1 is 0.6 and the toolchange retraction for that extruder is 17
[image]
[image]

### Expected results

A toolchange on the first layer should use the toolchange retraction length when swapping to the next tool. Think of the following use case:

1. A job using two extruders, T0 and T1, is about to finish. Both have significantly higher toolchange retraction lengths vs normal retraction lengths.
2. On the last toolchange going from T1 to T0, T1 retracts that higher value.
3. T0 finishes and the print job is finished. In this state, without a lot of extra end print g-code, T1's filament is retracted by a lot
4. The next print eventually starts where T0 is the initial tool so it prints the skirt and its portion of the layer
5. The first layer has a toolchange from T0 to T1. Instead of fully unretracting the toolchange length, only the normal retraction length is used for priming. This leaves the filament a long way from the melt zone.
6. T1 starts printing the first layer of its object but no filament is extruded due to the wrong amount of unretraction

Alternatively, being able to prime each tool right before it starts its portion on the first layer would be acceptable. Something like having having multiple tools print the skirt (I think Cura supports this, maybe PrusaSlicer, I'll have to dive back into those) or even a single layer prime tower (I don't want to enable a full prime tower). 

I currently have a custom prime sequence that extrudes a line for both toolheads, but the issue is that it's only simple to add before the print starts. So I prime T0 but then have to do a manual toolchange retract before priming T1 and doing the same. Since the initial_tool always handles the skirt, I simply prime manually prime the initial_tool before printing and hope any ooze comes off on the skirt.

### Project file & Debug log uploads

I don't think the log file is necessary since I'm not seeing a crash but please let me know if that would be needed for some reason and I can try to upload it. Project file is attached though.

[toolchange_retraction_issue.zip](https://github.com/user-attachments/files/17264727/toolchange_retraction_issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Please note that I have a custom wiper solution for my IDEX that doesn't use purge buckets (I have silicone that blocks the nozzle to prevent ooze). I don't want to prime while the extruder is sitting there because, as I stated above, that would lead to ooze as the extruder is traveling to the build plate and if it is not the initial tool, it is directly printing an object where I would want to avoid ooze (if I'm lucky I am using a brim but that's not viable all the time).

Please excuse any bad terminology/grammar and let me know if I need to clarify anything.

staal54a, 2025-01-04:
Yeah, I ended up working around it by creating custom prime lines for each extruder on my IDEX as part of the start g-code. Overall, I like that solution a little bit more in terms of end result, but the one problem is I run Klipper on that machine so I'm not using the START_PRINT macro at all. That just means I'm a little bit more locked into OrcaSlicer than I am on my other Klipper machine which only uses START_PRINT. 

So if I wanted to jump to Cura (which has been a while but the dual extruder support was okay from what I remember) I would have to port that start g-code over.

But yes, this issue should remain open (not entirely sure what causes the stale label to be removed) since it is a bug and would make toolchanger users' experiences a lot better once fixed.

Even something basic/crude like enforcing the skirt parameters for each extruder that print on the first layer (in terms of distance/material extruded) would work and be fine, albeit a little bit space inefficient.

Baggles5, 2025-01-04:
For the snapmaker with dual extruder, the startup G code has custom prime lines. The issue I have is that the custom prime lines prime both nozzles then performs a tool change retraction after priming the second nozzle before starting the print with the first nozzle. Then when ever the second nozzle is used (which can happen anytime) no de retract is performed on the first switch back to that nozzle. How did you fix this with your prime lines?

staal54a, 2025-01-04:
Yeah, so the part I didn't mention is that I basically do a similar thing but after the second nozzle's prime line finishes, it does do a toolchange retract but then after it parks, I manually prime it to while it is on the purge block (just a piece of silicone that blocks the nozzle). The end result of this is that instead of the filament being retracted toolchange_retract_length on that second nozzle, it is only retracted retract_length.

So on that first toolchange, the second nozzle gets properly primed when it travels to the print by simply doing a normal deretract which is what it currently does. I haven't really noticed much ooze with this method on the first toolchange and I think it would just result in a single layer of a slightly larger seam (since it comes it at a higher Z height and drops down onto the print).

The reason I opted for this is because there was no way for me to be able to move the second extruder/nozzle to its start point during the toolchange g-code. Because I did play around with manually priming the correct length on the first toolchange and I could only achieve something identical to my solution above where the nozzle is primed minus the normal retract length before it moves to the print. The way I did it allows my toolchange gcode to be pretty simple/consistent, at least for what it is.

Hope this all made sense, writing late at night makes me jumble my thoughts. I also will try editing parts of the bug so it remains open and not stale.

Baggles5, 2025-04-06:
This is still an open issue.

closed 2025-07-13 by github-actions[bot]