OrcaSlicer/OrcaSlicer #7019: OS does not change line width on nozzle diameter change when mm is used

WimVDK, 2024-10-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2 beta2

### Operating System (OS)

Windows

### OS Version

Win 11 23H2

### Printer

X1C

### How to reproduce

- Pick a nozzle diameter on printer eg 0.4
- have line width specified in mm (so 0.4 would be default in this case
- change profile to 0.6

### Actual results

Line width stays at 0.4mm with 0.6 nozzle

### Expected results

Coming from cura intially, when changing nozzle diameter, default line widths are automatically updated to match new nozzle size. In this case swapping from 0.4 to 0.6 nozzle would change the line width from 0.4 to 0.6 as well. Orcaslicer does not do this unless % is used. Seeing as its easier to read/calculate exact value (at least for me) to see actual numbers, i understand easier that 0.66 is a 10% increase than going 110% of 0.6mm

It would be nice to see orcaslicer automatically updating line widths when using mm

### Project file & Debug log uploads

none apply

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2024-10-11:
It did what you told it to do: Kept the line with at .04mm - That's not a bug.  You're just thinking about it wrong.
If you want the line width to automatically be the nozzle diameter then set the line with to 100%
> _its easier to read/calculate exact value (at least for me) to see actual numbers_
You having trouble with math is not a reason to break a feature many of use the way it is.
For example:
I have a profile set up that will do top lines at 100%, infil at 120%, exterior at 100% and interior at 110%
And I want the exact same behavior whether I run that profile on the machine with a .4 or a .6 or a .8 nozzle.

WimVDK, 2024-10-11:
Me having trouble with math isn't the issue. This is a UX problem and its not a profile issue either. It's a trivial thing to have orcaslicer automatically update layer widths just like cura can for my CR10

tlhintoq, 2024-10-12:
The program works correctly.  You're not willing to update how you work/think to use the new program.  If you want it to work exactly like Cura then use Cura.

You **have** a way of having Orca use the nozzle width of the selected printer: Choose `100%`.  Then you'll have a .4 line on a .4 nozzle and a .6 line on a .6 nozzle.  This is already how countless users use the software.  Breaking how everyone else share profiles between different printers because you don't want to use percentage instead of mm is ridiculous.

For example: Here we see two different printers, two different nozzle sizes (.6 on the X1c, .4 on the P1s), using the same profile perfectly well.  Insisting on not using the percentage ability means these two machines would have to duplicate profiles complicating a great deal.  There's just no need to force everyone else into keeping 10 profiles per nozzle size, per machine synchronized.
[image]

Your insistence on having everything in mm is _your_ issue and should not screw up the profiles and workflows of the thousands of other users.  You have the ability to do what you want to do, you just don't want to do it because you don't want to be burdened with math.  _"Oh, is this 100% of .4 or 100% of .6 - oh my, its too much math for me"_

> Me having trouble with math isn't the issue. 

### Well,  you said it was in your post.

> Seeing as its easier to read/calculate exact value (at least for me) to see actual numbers, i understand easier that 0.66 is a 10% increase than going 110% of 0.6mm

You're just _unwilling_ to use percentage.  That's on you not the software.  Its your issue, please don't try to turn it into a change that makes it everyone else's issue.

WimVDK, 2024-10-12:
I already AM using %, you don't know me or my settings so don't claim you do. It doesn't change that fact that a specific number being visible and automatically update when nozzle size changes, I'd easier to read

tlhintoq, 2024-10-12:
@WimVDK 
You're right - I don't know your settings - well, except for what you state your settings to be in the original post - Which you say are in MM.  Go back and read your own description of what you are doing and what you want, as if you were an outsider like me reading it.
You clearly state in several places that you are in MM and you want the values of line width to change to match the nozzle.
Your post title says "when mm is used".  The body of the post is "when mm is used".
So don't get upset with the reader for thinking you're using mm when everything you said to date is that you are using mm.  Its just a case of the reader taking you at your written word.
Now... several comments later... you have changed your tune to say you are using percentage.

But... Now moving forward... You used a new phrase that might be closer to what you're truly saying:
> a specific number **being visible**

So is it fair to say that you don't really want the line width to change to match the nozzle width - but instead it would be nice if a second number was shown as to what that width would be?  You say you're using percentage.  Ok.  So you have 150% entered same as in my screen shot.  You'd like there to be some kind of hint shown that since its a .4 nozzle the actual width will be .6mm - then a change to a different printer with a .6 nozzle - you're still at 150% actual value but the hint number changes to .9mm
Is this what you're describing?
[image]

WimVDK, 2024-10-12:
Do i really have to explain myself a third time? If the nozzle size is 0.2 i expect the default line with to state 0.2, if i change the nozzle to 0.4 i expect it to change to 0.4 (and similar values for the others), if i change it again to 0.6 or 0.8 i expect the default values to show 0.6 or 0.8.

Specific numbers are easier to read than percentages.

tlhintoq, 2024-10-13:
> Do i really have to explain myself a third time? If the nozzle size is 0.2 i expect the default line with to state 0.2, if i change the nozzle to 0.4 i expect it to change to 0.4 (and similar values for the others), if i change it again to 0.6 or 0.8 i expect the default values to show 0.6 or 0.8.

And that is exactly what happens when you set the default to 100% - AND YOU JUST GOT DONE SAYING YOU'RE USING PERCENTAGE - so what's the problem?

> Specific numbers are easier to read than percentages.
***FOR YOU*** - The thousands of other users are doing quite well with the system work as it is and as it has been.

A process profile such as the one in my example: `Armor` uses percentages and that single profile can be shared across 4 different machines, with 4 different nozzle sizes and work perfectly exactly the way its meant to.  I get 150% of the nozzle width on every machine regardless what is installed.  If I wanted it to be .6mm then I'd write it as .6mm - which is exactly what I do for this process of `Nice`.
[image]

What you are describing is the unwillingness to adapt to a new software - that's on you. You're arguing for your own inability to make use of MORE features and MORE options.  It is not incumbent on the writers to make 150,000 other users adapt to you wanting Orca to become Cura.

**You have a way to get exactly what you describe:** A default line width that is exactly the same as the nozzle width.  Its there... its baked in to the software, you enter a value of `100%` - _and you just got done saying you are using percentage_ - so why is this a problem?
[image]

tlhintoq, 2024-10-13:
I just thought of a different way of looking at this.  Maybe this workflow helps as to why to NOT change the default based on changing machines/nozzles.

Let's say I slice a job intentionally specifying a .6mm default line.
That's what I want the job to run at: .6mm line

- If I run that job on a machine with a .6 nozzle, then great - it will calculate extrusion at 100%, and I'll get my .6mm line.
- If I run the same job on a machine with a .4mm nozzle I still want a .6mm line.  _That's what I've told the profile I want:_ a .6mm line.  So the slicer will then do as its told and calculate a 150% extrusion rate so my .4 nozzle gives me what I've told it I want.

In other words - or looking at it from a different perspective - Orca expects the user to instruction the software about what it wants.  The user is in control.  The user tells the software what it wants and what to produce.  It does not alter the user's instructions or values. It does not tell the user what to expect just because there was a change in the user's target hardware.

You are telling the software:

1.  Give me a .6mm line
2. Use this machine
3. Now do your job to give me what I'm asking for. Don't change my settings. Don't tell me what I want. Don't be helpful and screw up my job {you aren't Microsoft}.  I'm telling you what I want, now do it.
And Orca happily complies with its orders.

closed 2025-01-20 by github-actions[bot]