OrcaSlicer/OrcaSlicer #7020: Extruder stops early in the print

SuperGripM, 2024-10-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

Windows

### OS Version

11

### Printer

TronXY X5SA-Pro w/Klipper v0.12.0-310-gef753468

### How to reproduce

During certain prints, the printer stops extruding at all, but moves are still done. Some prints work perfect, some don't. It always stops somewhere around 10-15 minutes into the print (somewhere past the first layer). I've not had issues with this before, nor with any smaller parts. I'm using adaptative pressure advance on the latest beta. GCode of a part that failed is attached.

### Actual results

Printer stops extruding.

### Expected results

Printer should extrude

### Project file & Debug log uploads

None available

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[1.6_PLA_1h10m.gcode_40.3272 g weight.zip](https://github.com/user-attachments/files/17266417/1.6_PLA_1h10m.gcode_40.3272.g.weight.zip)

vgdh, 2024-10-05:
in your gcode you have extrude moves up to the end.
Does the extruder gear turn during the print? May be it just clog.

SuperGripM, 2024-10-07:
Hiya,

Thanks for replying. The extruder stops moving completely, but if filament is manually pushed or fed from the gui (fluidd in my case) it works fine. No clogs on the hot end either.

I re-sliced the same part with adaptative PA disabled and it worked perfectly after failing twice with adaptative PA enabled.

vgdh, 2024-10-08:
can you provide your project file where issue occurs?

SuperGripM, 2024-10-10:
I'm going to try and make it fail, will report back. I don't have a habit of saving projects, so I'll try see what I can do.

SuperGripM, 2024-10-11:
Attached is a project file that failed just now.
[print test fail.zip](https://github.com/user-attachments/files/17341491/print.test.fail.zip)

vgdh, 2024-10-11:
Try to update clipper to latest version. Because Gcode was normal from the beginning.

vgdh, 2024-10-11:
I've experienced some similar behavior about a year ago when tried to print with too small steps, and after some time klipper just make a long reverse move on the extruder.

SuperGripM, 2024-10-12:
Thanks for looking into this, I will submit this to Klipper and see where we go from there. It's on the latest version.

SuperGripM, 2024-10-12:
Issue doesn't seem to be related to Orca after looking deeper into it.

closed 2024-10-12 by SuperGripM

zexcster, 2024-10-21:
no it's not a Klipper
I don't have klipper (KPS3 kingroon) and it does every time on this item
https://www.thingiverse.com/thing:3306294
on both 2.1.1 and 2.2.0 RC
I've fixed the model just n case, tried different settings.

frank111074, 2025-02-28:
The Problem is not Klipper! I've got the same problems with most Prints, sliced by Orcaslicer and Tronxy X5SA Pro.
If I slice my models with pepetier or Cura, the same model will be printet perfectly, sliced with Orcaslicer, Extruder stopps in most Models, and every File, in same height.
I tested it with Marlin 2.0.8 and Marlin 2.1.2.5 everytime the same.
BLOCK_BUFFER_SIZE raised from 16 to 32 makes no difference.

frank111074, 2025-03-02:
Yes, i saved the files, sliced with orcaslicer, with and without Preview jpg and a running cura gcode of the same part.
Even without the preview, the part will only printed ca. 1cm in height.
I had the same problems with a Model, that is ca. 27*30*37cm.
The extruder stopped at ca 1cm, not shure if it was exactlay the same height for each file, but anyway: files, sliced with cura are running as they should.

[X5SA_GPETG_Cartridge_Holder.txt](https://github.com/user-attachments/files/19044334/X5SA_GPETG_Cartridge_Holder.txt)
[Cartridge Holder_PETG_2h13m.txt](https://github.com/user-attachments/files/19044333/Cartridge.Holder_PETG_2h13m.txt)
[Cartridge Holder_PETG_2h11m.txt](https://github.com/user-attachments/files/19044335/Cartridge.Holder_PETG_2h11m.txt)

SuperGripM, 2025-03-09:
Apologies - should have closed this issue. Not related to Orca or Klipper, it was related to the stepper driver on the X5SA factory boards being configured to use StealthChop. Disabled the feature on hardware and the issue went away. Klipper/Orca implementing pressure advance and the X5SA having a long bowden tube aggravates the scenario, leading to an unrecoverable stepper driver shutdown after a certain time.
To people having this issue, especially on Tronxy printers - make sure this isn't your issue as well.