OrcaSlicer/OrcaSlicer #7026: Fan speed for overhangs depending on overhang degree, similar to overhang speed

Rickthebig, 2024-10-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Yes, I am frustrated when I find horizontal strips and bands in the printed piece if the appearance of material's surface is susceptible to cooling air (shiny type), typically at the height where overhangs exist.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I propose to change the hitherto existing cooling control for overhangs via threshold, which is to kick in at the set fan speed, since it is an all-or-nothing setting. Either the fan won't change its defined min. speed below this threshold or it will suddenly crank its speed up to the unique speed set for all overhang degrees above the threshold.

Instead, for several overhang degrees (25%, 50%, etc.) corresponding fan speeds would be defined, similar to the way overhang speed is set, and for interpolating all overhang degrees in-between.

This is to avoid sharp changes in fan speed which let the surface appear inconsistent.

Additionally, the fan speed should be modulated around the loop, i.e., return to its minimum defined value in parts of the perimeter without overhang, and gradually adapt its speed as the perimeter is built, depending on the overhang degree as set by the user. 

In other words, the fan speed should be specific to the current point along the perimeter. If one side of the part has an overhang, the fan should spin faster than on the side where the wall is vertically straight. Small perimeters won't benefit as much as large ones due to the sluggishness of the fan control, but that is still better than nothing.

As a result, the visibility of cooling artifacts will be minimized without compromising the overhangs.

### Describe alternatives you've considered

A simpler method for an adaptive fan speed control would be to define two values for the fan speed so that at any time the fan speed for an overhang would be an interpolated value between those. The lower value would be the minimal speed of the fan at the overhang threshold, the second one would be the maximum speed at (theoretical) 100% overhang. A linear relationship would be good enough.

chris99336, 2024-12-13:
I too would like to see this implemented. A single overhang angle threshold is clunky and leads to results that I have no control over. This is currently the only thing I want to see implemented that is a pain point for me. I love OrcaSlicer and I hope to see lots more added, but I think this one is necessary.

Felix14-v2, 2026-01-30:
I also think that having cooling interpolation would be really useful. Current approach just doesn't make sense in cases like this:

[image]

zipperzbieracz, 2026-09-20:
I support this request as I came up with that idea myself, facing some minor print issues. Now layer part cooling fan speed is variable only based on the layer time, but that's not enough for overhangs that are steep, but don't require supports, but are steep enough that the outside layer does a mini-curl due to insufficient cooling. I can increase minimum fan speed, but that would blow harder on the entire print thus weakening it, and not in the places it precisely needs just a little extra cooling so the outside surface comes out perfect, without compromising print strength. Now, as @Felix14-v2 posted in the photo, the only option for it is the overhang angle cooling, but that's either 100% or the minimum fan speed, and that can create problems on it's own. User should be able to create a gradient between some safe X% overhang angle (minimum cooling as in the preset) and Y% overhang angle (which would use maximum cooling for short layers or the 100% overhang cooling activation threshold). Just like slicer creates a cooling gradient with the layer time. This would be a toggleable feature for expert users.