OrcaSlicer/OrcaSlicer #7028: Qidi plus 4 profile

camrsnow, 2024-10-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am frustrated not being able to use my plus 4 with orca

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I'd like to be able to use orca with my plus 4

### Describe alternatives you've considered

I have the default slicer, but I'd like to use orca

seris007r, 2024-10-06:
i second this request, it would be fantastic to have a plus 4 profile added for the new qidi device.

https://qidi3d.com/products/plus4-3d-printer

noahkatz, 2024-10-07:
The latest version of orchestra has the Plus 4 listed.

Blair1999, 2024-10-16:
Thirded! My plus 4 will be here Friday... :)

closed 2025-01-23 by github-actions[bot]