OrcaSlicer/OrcaSlicer #7029: Seems so obvious, using the Ext spool while also using AMS

mytechguyri, 2024-10-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

This pertains to the Bambu line of printers.  I have need to print models that are PLA or PETG, and also have some TPU sections, or possibly even multi-color TPU (phone cases immediately come to mind), Of course when using the AMS, the software says you can only use filaments loaded in the AMS, and of course you can't use TPU with the AMS....  

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

The solution seems so obvious, I cannot understand why it wouldn't be implemented... as I believe to accomplish this would only require sending the appropriate g code to the printer.   So lets suppose I have 4 different colors of PLA loaded into the AMS... and I've got a spool of TPU sitting on the ext spool rack... I want to print a model that have both PLA and TPU in the model, or perhaps I want to overprint TPU over the PLA... Why can we not handle all the PLA printing with the AMS as normal, but when TPU is called for, which is noted in the slicer and on the printer settings as on the Ext Spool, so the printer knows the TPU filament is on the external spool, send the gcode to unload the filament from the AMS, just like would happen for a color change... but then prompt the user to manually load in the filament from the external spool... the printer can then print the TPU for that layer, when done, again, manually prompt the user to unload the external spool filament, and continue printing with the AMS as normal until TPU is needed again.... now, understood this makes it a completely manual, and likely tedious process.... but it would at least give the user the option to do it if they so choose... The model I'm working on, it prints a graphic logo on the first layer in 4 colors in PLA, but then the entire rest of the model is TPU... so this would basically be just one manual load/unload, and gives the end user significantly more options.

### Describe alternatives you've considered

printing two separate models, one with the PLA, then printing the TPU sections as a separate model, without cleaning the build plate between the prints.... but this would be a nightmare to get everything to come together correctly.

### Additional context

basically.... if filament needed is configured on external spool, unload filament from the AMS, prompt user to load in filament on external spool, print, prompt user to unload from external spool, continue with AMS as needed.

Syclusion, 2024-10-06:
No offense, that sounds like hell as an end user.

Syclusion, 2024-10-06:
May I reccomend an AMS upgrade instead? Hydra is a good place to start

mytechguyri, 2024-10-07:
> May I reccomend an AMS upgrade instead? Hydra is a good place to start

Hydra STILL doesn't make it able to handle TPU.

AbeFM, 2024-10-07:
I largely agree with the OP, and have considered manual sharing between external and AMS spools. I think it is a good idea, but not the most direct to MMU prints using the AMS with TPU.
Instead, make the AMS work and have the process automated. https://github.com/SoftFever/OrcaSlicer/issues/7045

mytechguyri, 2024-10-18:
> I largely agree with the OP, and have considered manual sharing between external and AMS spools. I think it is a good idea, but not the most direct to MMU prints using the AMS with TPU. Instead, make the AMS work and have the process automated. #7045

There are simple ways to make the AMS allow you to print TPU.   The simplest of which is, simply edit your C:\Program Files\Bambu Studio\resources\printers\filaments_blacklist.json file and set the action for tpu to "warning" instead.  Then you can select TPU for the AMS, and it will just pop up a warning, but still allow you to proceed instead of an outright prohibition on using TPU.

Here's what my filaments_blacklist.json file looks like after editing.
`"whitelist": [
  ],
  "blacklist": [
    {
      "vendor": "Third Party",
      "type": "TPU",
      "action": "warning",
      "description": "TPU: not supported"
    },
    {
      "vendor": "Bambulab",
      "type": "TPU",
      "action": "warning",
      "description": "TPU: not supported"
    },
    {
      "vendor": "Third Party",
      "type": "PVA",
      "action": "warning",
      "description": "PVA: flexible"
    },
    {
      "vendor": "Third Party",
      "type": "PLA-CF",
      "action": "warning",
      "description": "CF/GF: hard and brittle"
    },
    {
      "vendor": "Third Party",
      "type": "PETG-CF",
      "action": "warning",
      "description": "CF/GF: hard and brittle"
    },
    {
      "vendor": "Third Party",
      "type": "PA-CF",
      "action": "warning",
      "description": "CF/GF: hard and brittle"
    },
    {
      "vendor": "Third Party",
      "type": "PAHT-CF",
      "action": "warning",
      "description": "CF/GF: hard and brittle"
    },
    {
      "vendor": "Bambulab",
      "type": "PET-CF",
      "action": "warning",
      "description": "Bambu PET-CF/PA6-CF: not supported"
    },
    {
      "vendor": "Bambulab",
      "type": "PA6-CF",
      "action": "warning",
      "description": "Bambu PET-CF/PA6-CF: not supported"
    }
  ]
}
`

While my AMS will handle 98A TPU without issue, it chokes too much on 95A or softer.... so you still need an option to print from the external spool while using the AMS

closed 2025-01-25 by github-actions[bot]