OrcaSlicer/OrcaSlicer #7030: Add filament preset for ABS-GF

WimVDK, 2024-10-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

yes. Under custom filaments > generic filaments, no "abs-gf" preset exists

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add a preset for abs-gf under generic filaments

### Describe alternatives you've considered

there are no alternatives

dMopp, 2024-10-06:
Bambu ASA-CF preset can be found in bambustudio already, even if its not in resources in github ...

dMopp, 2024-10-06:
[Copy ASA-CF of Original.zip](https://github.com/user-attachments/files/17271135/Copy.ASA-CF.of.Original.zip)
Here is a filament copy

WimVDK, 2024-10-07:
I'm not talking about Bambu. I explicitly said GENERIC filament

closed 2025-01-13 by github-actions[bot]

thetic, 2025-05-01:
there is still no ASA-GF support. Please reopen this ticket.