OrcaSlicer/OrcaSlicer #7045: Allow TPU in AMS

AbeFM, 2024-10-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I want to load TPU into the AMS, which works absolutely fine and completes prints more often than BBL PETg did.
You can skip the blacklist on the slicer, but unless you lie to the printer, telling it your TPU isn't TPU, it will not print.

Lying has secondary effects, such as future prints will auto-select your TPU and try to print a PLA print at high speed, stressing or breaking the hardware. see further detail here: https://github.com/bambulab/BambuStudio/issues/2096

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Ideally, a window pops up saying "you can't use this filament unless you understand the risks", but either way, loading TPU should just work. In an ideal world, we can take advantage of knowing we're loading TPU by loading it slowly, etc.
Also, differentiating between the > 90A stuff and soft stuff would make a lot of sense. I might not run NinjaFlex through the AMS, but most off-the-shelf cheapy stuff works great.

### Describe alternatives you've considered

Begging the Bambu Labs team for years  https://github.com/bambulab/BambuStudio/issues/2096#issuecomment-2395652680
Posting pictures and profiles of multi-color TPU and multi-material-including-TPU prints about the internet
Building a time machine and going back to when this used to work as the Kickstarter machines did this when I got mine
Bringing it up here

### Additional context

I haven't tried Daniel's code, but the idea sounds right, and the advanced users of the Orca community would be the right folks to see how it goes in practice. His code is here: https://github.com/daniel-brenot/BambuStudio

kylek29, 2024-10-13:
> differentiating between the > 90A [filament stiffness]

That might be best implemented as a filament profile setting that users could then account for (e.g. with custom GCode to slow it down or what-not).

phiten, 2024-10-18:
I fixed that in this PR https://github.com/SoftFever/OrcaSlicer/pull/7161 

Installers and stuff can be found in my repos actions https://github.com/phiten/OrcaSlicer/actions/runs/11401874109

Basically the PR checks if the checkbox is set and changes prohibited filaments to filament with warning so that the confirm button actually sends the print.

phiten, 2024-10-23:
@SoftFever I think this issue can be closed with the merge of my PR #7161 

There is no differentiation between the hardness but I think thats fine. In my testing with the AMS lite it just won't work but it doesn't break stuff. And the warning window also still pops up.

closed 2025-01-30 by github-actions[bot]