OrcaSlicer/OrcaSlicer #7051: SEAMS OPTION

Noname987654, 2024-10-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The SEAMS settings are very frustrating  in comparison with CURA which allows a lot more control with its ALIGNED option
 (left, right, back, back left, back right, front, front left, front right).

For example, if you print multiple same objects, and need to turn some in 180° rotation, the BACK for some becomes the FRONT for others, and since we don't have FRONT, we can't produce the exact same print regarding their orientation on the build plate.

Yes we can paint on the object, but in the case presented it's an insufficient hack.

THX for all the work ! 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

more control with its ALIGNED option
 (left, right, back, back left, back right, front, front left, front right).

tlhintoq, 2024-10-12:
If you paint the seam in place on the first object... Then clone and rotate as needed... Doesn't the seam stay in the same place relative to the part?

kylek29, 2024-10-13:
> If you paint the seam in place on the first object... Then clone and rotate as needed... Doesn't the seam stay in the same place relative to the part?

I think he's talking more about the automated seams generation. The system seems to align them orientated to the plate/3d-space, not the model's orientation.

Ideally, the same model shape using the same settings for each clone should be identical (outside of exterior forces like the fan air direction) regardless of their rotational orientation on the plate.

tlhintoq, 2024-10-13:
Oh for sure he's talking about the automated seams.  But come on, who really relies on those, past the first month of 3d printing?  Certainly not someone that has moved forward in to doing batches of the same item large enough to fill the plate and require re-orientation to fit more.  If you're doing batch production you should be past the rookie stage of just accepting the automated seams.

Either way he's got to accept that `Back` is back of plate not back of object.  That's how it is in this slicer and the slicers it inherits from.  There's two choices:
1 - Stop printing until someone decides to change a very fundamental concept of the program.  Which won't happen because everyone else already using it the way its written will be upset.  The developers are not going to change foundational items like this and piss off the existing user base.
2 - Find a way to adapt.   And there is a way to do that *today* by moving past automated seams and taking 60 seconds to paint them in on your first object _then_ clone.

Noname987654, 2024-10-16:
> Oh for sure he's talking about the automated seams. But come on, who really relies on those, past the first month of 3d printing? Certainly not someone that has moved forward in to doing batches of [...]

> > LOL What a clueless condescendent assertion  ...  
Though, reading people embarrassing themselves is sometimes funny. 

> Either way he's got to accept that `Back` is back of plate not back of object. That's how it is in this slicer and stop trying to make Orca into Cura. If you want Cura, use it. There's two choices: 1 - Stop printing until someone decides to change a very fundamental concept of the program. Which won't happen because everyone else already using it the way its written will be upset. The developers are not going to change foundational items like this and piss off the existing user base. 2 - Find a way to adapt. And there is a way to do that _today_ by moving past automated seams and taking 60 seconds to paint them in on your first object _then_ clone.

>>  That's how it is in this slicer and stop trying to make Orca into Cura

So you would probably want to remove many options in Orca existing thanks to other slicers like Cura ? 
Btw, Orca Slicer is based on Bambulab Slicer which is based on Prusa Slicer, which is based ..on Slic3r .. ROLF
And I might be wrong, but first implementation of : Fuzzy Skin : Cura ; Vase mode : Cura ; Arachné : Cura  ; .... 

Dude, when you have 14 yrs old, just discovered 3D printing 2 or 3 yrs ago, and just talk nonsense, then you shouldn't even be writing and just keep your garbage for yourself. 
But you surely made a lot of people laugh their ass off ! Good job on that though !

tlhintoq, 2024-10-16:
Congrats on having a 14  year old.  I have a 30 yo and 32 yo.  Kids (and grand kids) are a joy.
Its terrific to pass along the hobbies you enjoy to them.  Cool that you've been doing this a couple years.
I've been doing this for a living since 2009 - so my opinion that if you've gained enough experience to be batching large quantities of the same part usually comes along with enough experience to paint in your own seams and supports on the first one then clone isn't just being condescending.  Its based on about a decade and half of full-time experience in the shop and interacting with countless other shops and individuals.  It's not an insult to you so please don't get riled up.  I'm just saying that if you reach a point to do a certain thing you generally build up other related skills to go with it.  Kind of like cars: If you are talking about rebuilding the transmission you generally have enough experience to know how to change the socket on your socket wrench and probably know how to use a torque wrench.  The skills kind of build up in parallel as you progress through the hobby.  Same here with getting far enough to running batch jobs probably goes along with no longer relying on auto supports and auto seams.

Noname987654, 2024-10-16:
"Congrats on having a 14 year old." 

You can't even read, this is very embarrassing... 

More polluting garbage from you ... 

You might think you're on reddit or Twitter - a common issue with braindead folks like you, I pity your family -  but this is actually Github, (we've seen that you can't read, so, we understand your misconception) so if you're not contributing to the thread, you can ridicule yourself elsewhere.

tlhintoq, 2024-10-16:
Much of your post reads like English is a second language.  So forgive me if I didn't translate this the way you intended.  But it does read like you're saying that you have a 14 year old, that has being doing 3d printing for the last 2 to 3 years.  Its your words.  Don't get angry at someone taking it the way your wrote it.
> Dude, when you have 14 yrs old, just discovered 3D printing 2 or 3 yrs ago

At this point you're getting hostile, insulting and abusive.  Nobody here has name called you, insulted you, or anything similar.  You're escalating and making things very personal.  I invite you to re-read the thread and realize that you're taking this in a direction you probably don't want to go.

Noname987654, 2024-10-17:
> Much of your post reads like English is a second language. So forgive me if I didn't translate this the way you intended. But it does read like you're saying that you have a 14 year old, that has [...]
How Funny is it that you didn't quote the end of the sentence, which is quite telling about either your dishonesty or your inability to read : 
Dude, when you have 14 yrs old, just discovered 3D printing 2 or 3 yrs ago, **AND just talk nonsense, THEN YOU shouldn't even be writing and just keep YOUR garbage for yourself.**

> At this point you're getting hostile, insulting and abusive. Nobody here has name called you, insulted you, or anything similar. You're escalating and making things very personal. I invite you to re-read the thread and realize that you're taking this in a direction you probably don't want to go.

Talking about hostile, insulting and abusive ? Oh let's read you again : 

" But come on, who really relies on those, past the first month of 3d printing? Certainly not someone that has moved forward in to doing batches of the same item large enough to fill the plate and require re-orientation to fit more. If you're doing batch production you should be past the rookie stage of just accepting the automated seams."

But again, all the stupidity and nonsense that you've said is here for the amusement of everyone to read.

closed 2025-01-24 by github-actions[bot]