OrcaSlicer/OrcaSlicer #7052: See Photo, Excess plastic from extruder bug on multiple levels from Arachne.

ZoomDomain, 2024-10-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2..2

### Operating System (OS)

Windows

### OS Version

win 10

### Additional system information

Please see photo, It's form Orca 2.1 or 2.2???
first error layer is at about 19.00, then 23.80, 25.85, 28.2, they look like they coincide with tiny overhangs in the cleft-joint on the X+ side of the cylinder. 

[image]

### Printer

elegoo neptune pro 4

### How to reproduce

just run the gcode with given parameters and print. 

### Actual results

multiple gcode layers had errors. 

### Expected results

to have no errors on the plastic

### Project file & Debug log uploads

[multiple faulty layers of excess extrusion on outer gear only.zip](https://github.com/user-attachments/files/17297632/multiple.faulty.layers.of.excess.extrusion.on.outer.gear.only.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[freewheel sun and output ring x axis panetary.zip](https://github.com/user-attachments/files/17297642/freewheel.sun.and.output.ring.x.axis.panetary.zip)

ZoomDomain, 2024-10-08:
The general result from arachne was rather bad, there was a lot of travel for the dots of the inner wall and the travel was randomly inefficient. Here is the Gcode. The extrusion ratio is wrong at region z=19.00, then z=23.80, 25.85, 28.2. I don't know if it's 2.1 or 2.2, I think it was 2.2.beta.

Ergonomicmike, 2024-10-08:
FWIW, I opened your project in Orca 2.2.0-rc. I do not see any changes in flow, or speed, or layer height, at any z regions in your model. The preview looks good to me. (Except that I don't understand the sideway-V split in the model. Apparently part of the model?)

I do not know why it should be different for me than for you. (I did not print the gear for a number of reasons, one being that I don't have ABS. Nor do I have 16 hours.)

ZoomDomain, 2024-10-09:
> FWIW, I opened your project in Orca 2.2.0-rc. 

Thanks a lot for the tips, sorry I forgot to include the gcode, I tried to read the gcode just now, i.e. layer 19.094, It can be complicated for me to find the ratio of extruder and displacement speed to see if it is differerent between Z values for the outer wall. I'll try to figure a way to see where the gcode artefact is.

Ergonomicmike, 2024-10-09:
I exported the gcode from Orca and inspected it using the latest version of Prusa's g-code viewer.

I don't see any of the reported errors in the viewer. (No change in flow, layer width, speed, etc.)

closed 2025-01-16 by github-actions[bot]