OrcaSlicer/OrcaSlicer #7059: Allow use of G0 instead of G1 for travel moves.

ADKMechETECH, 2024-10-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-Beta

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

### Printer

Raise3d Pro3

### How to reproduce

1. Slice any file. 
2. View G-code and observe all travel moves use G1 X## Y## F### format. 
3. Run code on Raise3d Pro3 - Observe travel moves are not respecting F### from the G1 move. 

### Actual results

G0 Works perfectly fine in the start G-code I have modified to make the printer use its rapid speed.  G1 commands appear to occasionally ignore the F### value for rapid moves.  

### Expected results

I do not know why the G-code parser of the Pro3 is behaving like this, but the easy solution is to have a toggle in the machine config to enforce using G0 instead of G1 for all travel moves. 

### Project file & Debug log uploads

[arms.zip](https://github.com/user-attachments/files/17317183/arms.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Ergonomicmike, 2024-10-09:
This Issue appears to me to be a Request for Enhancement as opposed to a bug report. Orca isn't doing anything wrong.

ADKMechETECH, 2024-10-10:
Sorry, not an experienced Github user here.  I'll post in that section. 

Closing as this is not actually a bug, it's a feature... .lol

closed 2024-10-10 by ADKMechETECH