OrcaSlicer/OrcaSlicer #7061: Wipe randomly disabled with Avoid Crossing walls

Travis90x, 2024-10-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10 22H2

### Additional system information

Ryzen 4600G + 32GB RAM + SSD M.2

### Printer

Bambulab A1 + AMS

### How to reproduce

using a hollow sphere, wipe distance 1mm + wipe while retracting

### Actual results

the wipe randomly is skipped when avoid crossing wall is enabled

### Expected results

the wipe may still the same if avoid crossing wall is enabled

### Project file & Debug log uploads

[Sfera cava.zip](https://github.com/user-attachments/files/17319782/Sfera.cava.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]
[image]

discip, 2024-10-10:
Please test the latest build, as this issue was already addressed!

Travis90x, 2024-10-10:
> Please test the latest build, as this issue was already addressed!

Thanks, 2.2.0 RC seems to work fine!

closed 2024-10-10 by Travis90x