OrcaSlicer/OrcaSlicer #7064: Floating 'overhang walls' with inner/outer/inner wall order

3LD, 2024-10-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

SV06 Plus

### How to reproduce

Set 'wall printing order' to 'Inner/Outer/Inner'. 
Slice up an upside down box shape. 

### Actual results

Attempts to use overhang printing settings to print out in thin air. 
Makes spaghetti 

### Expected results

less/no spaghetti 

### Project file & Debug log uploads

[spaghet.zip](https://github.com/user-attachments/files/17320934/spaghet.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

vgdh, 2024-10-10:
also related https://github.com/SoftFever/OrcaSlicer/issues/6662

kylek29, 2024-10-13:
This also occurs with other shapes as well (such as a round cap). I've had a few failed prints because I didn't catch it in slice preview, now I've learned to check for it specifically if I think there's a chance at a certain layer.

closed 2025-01-20 by github-actions[bot]