OrcaSlicer/OrcaSlicer #7066: Multimterial Printing Order

dMopp, 2024-10-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I tested a Multi material print (Print by Object) and figured out, it would be a big improvement (for energy usage and/or quality) if the print order would based on the Bed Temperature. During my print with ASA, PLA and PETG the bed was turning up (thats fast) and colling down (very slow) multiple times which might cause adhesion issues or even melting PLA first layer. So for my POV the best solution would be starting with the highest Bed temperature and after changing it (lower it) the printer should wait until the target temp is reached and not starting printing to fast (colling takes more time then filament change)

### Additional context

Iam using a Bambu P1S with AMS

tlhintoq, 2024-10-11:
Since you control the print order, and you decide what object is what material, can't you just print them in the order you want them?
Don't add more plumbing to break, just think for yourself and tell it to do what you want.

dMopp, 2024-10-11:
How do I control the print order myself? Iam not aware of that. For sure that would be enough! (But the printer should wait longer for cooldown anyway)

tlhintoq, 2024-10-11:
View the list of objects: That's the print order
(Flip the switch from `Global` to `Object`
Move them up and down in the list to change the order
[image]

closed 2025-01-18 by github-actions[bot]