OrcaSlicer/OrcaSlicer #7070: First layer on top of supports should not be seen as a bridge

jamincollins, 2024-10-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-rc

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Printer

RR v-core 3.1 400mm

### How to reproduce

use any model with long unsupported or cantilevered section
enable supports
slice model
check first layer just above supports

### Actual results

layer just above supports is treated as a bridge

### Expected results

should not be treated as a bridge as it is _supported_

### Project file & Debug log uploads

[bridging-issue.3mf.zip](https://github.com/user-attachments/files/17333742/bridging-issue.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

sleewok, 2024-11-14:
Personally I'm not sure I would want this changed and I don't think it is a bug. The layer is "floating" since there is a gap between the support and following layer. I just did a test print with a support setup on a bridge that isn't being detected properly as a bridge. It attempted to print it at full speed and the result was not good. The layer did not extrude properly. Based on my result with it I would not recommend making this change. At most have a toggle (or individual speed setting), but I don't think it should be default.

jamincollins, 2024-11-14:
The problem isn't so much with speed. For me it was cooling. Bridges tend to want high cooling fans, printing on top of supports the cooling fan can (and does) result in blow back disturbing the layer being laid down.

closed 2025-02-20 by github-actions[bot]