OrcaSlicer/OrcaSlicer #7072: Fan Forced 100% For 3 Layers

CCS86, 2024-10-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 RC

### Operating System (OS)

Windows

### OS Version

10 x64

### Printer

X1C

### How to reproduce

...

### Actual results

...

### Expected results

[image]

### Project file & Debug log uploads

[Fan speed.zip](https://github.com/user-attachments/files/17336760/Fan.speed.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

kybarg, 2024-10-11:
@CCS86 seems like this is related to "Fan speedup time" setting.

hagre, 2024-12-16:
I’m experiencing something similar in release version 2.2.

I have spitted a part horizontal, just to use a different color (same material) for the top layers.

[image]

 The fan speed for the top layers was not as expected!

[image]

 Orca is using the overhang-fan-speed of the bottom material for the flowing layers.

[image]

hagre, 2024-12-16:
Folow up:

If you deselect the "enable_overhang_bridge_fan" everything is as expected. Looks like the top part is handled like an overhang even if it is practical not (parts touching each other).

[image]

[image]

closed 2025-03-25 by github-actions[bot]

hagre, 2025-03-25:
Still the same at 2.3.0.0