OrcaSlicer/OrcaSlicer #7073: On multiple multi color prints the prime tower is super long/Gigantic. Goes off the build plate or is higher then the print itself.

SummonSign, 2024-10-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambu Lab A1

### How to reproduce

1. select a multi color print from makerworld
2. set up the prepare settings 
3. go to the preview tab

### Actual results

Orca Slicer spawn in a giagantic or really really long prime tower. usually longer then the print bed or it takes up half the plate. 
[image]
[image]
[image]
[image]

### Expected results

it should be a small prime tower, or at least as long as the print bed... 

### Project file & Debug log uploads

[SunLu+S4+accessory+box+3mf.zip](https://github.com/user-attachments/files/17337390/SunLu%2BS4%2Baccessory%2Bbox%2B3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

someone on discord said for me to fix it, close orca and reopen it, for the small box it fixed it but for a majority of other projects its always spawned in gigantic.

vgdh, 2024-10-11:
I agree it confuses when you see it the first time.
It happens because of too high flushing values.

vanowm, 2024-12-16:
> I agree it confuses when you see it the first time. It happens because of too high flushing values. You can decrease them manually.

Unfortunately, flushing values have no effect on the tower whatsoever.
Testing in v2.3.0-dev on Windows 10 x64

closed 2025-03-25 by github-actions[bot]

vanowm, 2025-03-25:
Good job, closing a bug report as "not planned" is the way to go! Why wait, just close issues as soon as they created...