OrcaSlicer/OrcaSlicer #7077: Incorrect infill generation with Ensure Vertical Shell Thickness

Nemernemer, 2024-10-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11 (Atlas OS)

### Printer

Custom Voron 2.4

### How to reproduce

Import any model that has a vertical wall adjacent to a horizontal surface.
Slice with "Ensure vertical shell thickness" set to anything other than "All"

### Actual results

The last layer for the adjacent top surface incorrectly skews the walls for the next layer in, filling the new space between shell and solid infill with sparse infill.
[image]

This causes a very weak layer bond on these intersections, and even gaps in the printed model.

### Expected results

I would expect the same result as with the setting on "All":
[image]

### Project file & Debug log uploads

[Assembly.zip](https://github.com/user-attachments/files/17346562/Assembly.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

This is not a recent issue for me, but I only recently discovered why it was happening.
Repeats on multiple installs between different computers (all Win11).

Nemernemer, 2024-10-18:
I would have to disagree on strength not being significantly affected.
In the case of staggered features, such as a cylinder sticking out of a flat surface, the entire protruding feature is only bonded to the previous layers through sparse infill:
[image]

Had a couple parts like this break, which led to me investigating what was causing this defect to begin with.

xokahag, 2024-10-19:
Because *most* models print just fine, there is not more people besides us complaining about this issue.

I posted screenshot exactly because it is a real issue, and it needs the attention to get it fixed.

chgi, 2024-12-21:
Just want to chime in with a picture of an actual print. This causes clearly visible ugly gaps between top surface and wall.
This is with version 2.2.0 under Linux.

[image]

Also, thank you @Nemernemer for pointing out this is caused by _Ensure vertical shell thickness_.
I was kicking myself, because I couldn't find out what caused this.

otede, 2025-03-22:
Does anyone have any recent examples or a link to a model I could try myself? EVST has been problematic for me since my beginnings with Orca a year ago - the feature being overzealous where it's not needed and underperforming in critical areas where theoretically it should have saved the day.

closed 2025-10-01 by github-actions[bot]