OrcaSlicer/OrcaSlicer #7079: Bed Temperature changes when using multiple tools

lukasschwoerer, 2024-10-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

Not relevant

### Printer

E3D Toolchanger

### How to reproduce

Create a multi tool print job using filaments with different bed temperatures.

### Actual results

The printer will change the bed temperature according to the filament currently used. For a job using PLA and PETG it tries to warm/cool the bed from for excample 60 to 80°C and back for every layer.

### Expected results

What I would expect is that the slicer just picks the highest bed temperature used and prints everything with that temperature. It makes no sense to cool down the print plate mid-print because this causes the higher temperature filament to release from the print plate. 

### Project file & Debug log uploads

[ExcampleMultiToolJob.zip](https://github.com/user-attachments/files/17347828/ExcampleMultiToolJob.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ergonomicmike, 2024-10-11:
I am hoping to buy an IDEX soonish. So I have no experience with multi-material printing yet. But what I would expect is that the bed temp is set by the first layer and stay there, since that's the layer/material that needs to stick.

lukasschwoerer, 2024-10-12:
> I am hoping to buy an IDEX soonish. So I have no experience with multi-material printing yet. But what I would expect is that the bed temp is set by the first layer and stay there, since that's the layer/material that needs to stick.

Yes, more or less. Having the bed 5°C hotter for the first layer can sometimes help to achieve better adhesion, but after that it should stay constant.

Ergonomicmike, 2024-10-12:
Now that I've "slept on it," I can see a case where I might have my support material in PLA and my main print material in PETG. In that case, I would want the bed at 80 degrees C for the PETG but only 65 for the PLA supports.

I dunno... maybe an Option to override changing bed temp and either assign it a fixed value or chose which material wins in the fight over bed temp? (Just brainstorming.)

staal54a, 2024-10-13:
Yeah, it's an interesting situation. The problem with going with the highest temperature is what if the case was you only had PLA touching the build plate but PETG on other parts of the print?

I have an IDEX I am finishing up a few mods on and eventually I would want to do multimaterial prints as well (currently just dual PLA). I've thought about this issue for a bit but I haven't been able to come up with a good solution that covers all cases. 

The best option is probably some type of dropdown/config menu that allows you to select using the bed temp from one of the tools, highest temperature or change it each time. It would still be the responsibility of the user to make sure it was configured correctly for that specific print but would make it very simple to do.

lukasschwoerer, 2024-10-13:
> Yeah, it's an interesting situation. The problem with going with the highest temperature is what if the case was you only had PLA touching the build plate but PETG on other parts of the print?
>
> I [...]

True! that's a case i havnt thought about, and you are absolutely right. In the case where only the support material touches the build plate, we want the build plate to be at the correct temperature for that material.
Maybe some logic could be implemented that looks at the first layer and determines the correct bed temperature based on that.

closed 2025-01-20 by github-actions[bot]