OrcaSlicer/OrcaSlicer #7080: Add Z axis hole and contour compensation

DecVes, 2024-10-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Im trying to print a Flight stick for a flight simulator and its very technical, I already knew my 3d printer was off by a bit on the X and Y axes for contours and holes. So I usually put my desired amount in to the existing X and Y compensation and it works like a charm but for prints that im doing recently (especially longer prints) it will be a pain to go back into blender and mess with my models for them to even fit correctly.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would love for the Z axis compensation for both holes and contours to either be separate or built in to the X and Y compensation 

### Describe alternatives you've considered

As I said before, it will be a pain to go back into blender just to make the models fit.

kylek29, 2024-10-13:
Maybe a form of dropdown to introduce an automated form of horizontal hole compensation? There's a few mentioned [here](https://youtu.be/Bd7Yyn61XWQ?t=1m20s) specifically, the teardrop method. The hole is modified to add a tangent triangle that takes out the bridge, preventing the droop/flat top of the hole.

It works really well but adds a bunch of extra work to the 3d Model that you don't necessarily want in your master model, so it'd be better done on the slicing side.

I mention a drop-down, because I'm sure there's some other techniques people have, so it'd work best as something where the end user can select their desired method.

DecVes, 2024-10-13:
> Maybe a form of dropdown to introduce an automated form of horizontal hole compensation? There's a few mentioned [here](https://youtu.be/Bd7Yyn61XWQ?t=1m20s) specifically, the teardrop method. The hole is modified to add a tangent triangle that takes out the bridge, preventing the droop/flat top of the hole.
> 
> It works really well but adds a bunch of extra work to the 3d Model that you don't necessarily want in your master model, so it'd be better done on the slicing side.
> 
> I mention a drop-down, because I'm sure there's some other techniques people have, so it'd work best as something where the end user can select their desired method.
> 
> Snapshot: ![Screenshot_20241013-134741](https://private-user-images.githubusercontent.com/15932180/376068442-6363fa7d-6076-40a2-aba6-e6ce6fc0406e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg4NTI4NTgsIm5iZiI6MTcyODg1MjU1OCwicGF0aCI6Ii8xNTkzMjE4MC8zNzYwNjg0NDItNjM2M2ZhN2QtNjA3Ni00MGEyLWFiYTYtZTZjZTZmYzA0MDZlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEzVDIwNDkxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUzNzYzMjMwZjhjMTY2YWY5MWZjNjY4M2M5MmY3ZjJlZjYyZGRjMjg2OWYxODUxYmM5YTZiZDNlZWRlOGUwMjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.struqY0S_CcZIJ0HP7yAGu9SN_H3coCeeHzGZ4N1Z-E)

I didn't know that was an option, that's actually pretty neat, I would still love for them to automate the horizontal holes and vertical holes but that is a nice workaround

kylek29, 2024-10-13:
> I didn't know that was an option, that's actually pretty neat, I would still love for them to automate the horizontal holes and vertical holes but that is a nice workaround

I'm in agreement that it'd be great as a slicer option. It does add a bunch of extra work.

Another option would be to add the ability to "add extra walls around holes" (both vertical and horizontal -- horizontal would be more fill layers), this gives you sacrificial plastic (without compromising the hole) to use a precision drillbit (or for heat inserts).

You _can_ design that in by making two models, one with the just the "hole walls" (floating in space) and the main model with the holes as desired, the "hole walls" version would be inserted as a modifier with the tweaked parameters.

closed 2025-01-20 by github-actions[bot]