OrcaSlicer/OrcaSlicer #7083: Adaptive Purge Line

kekkodance, 2024-10-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Like shown in Teaching Tech's latest video, adaptive purge lines allow for less stringing and an easier removal. My current worries are that if you have a full plate model, the purge line would be outside of the boundary and thus Orca would need to revert to the stock purge line.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Implementing adaptive purge lines that adapt even if the model is the max area of the plate.

brjohnso1, 2024-10-12:
Somebody must be doing something, because 2.2RC is scribbling purge lines out of bounds.

brjohnso1, 2024-10-12:
[image]
WTF

tlhintoq, 2024-10-13:
2.2-RC
Looks fine to me
[image]

kekkodance, 2024-10-14:
(this issue was for something else and you guys turned it into a bug report btw)

tlhintoq, 2024-10-14:
> (this issue was for something else and you guys turned it into a bug report btw)

Its not tagged as a bug.  Its tagged as an `enhancement`.  IE: Feature request
[image]

closed 2025-02-15 by github-actions[bot]