OrcaSlicer/OrcaSlicer #7085: Skip Ender 3 S1 Pro filament blacklist

chomiczon, 2024-10-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Yes it is a problem. Ender 3 S1 Pro can print blacklisted filaments, for example TPU

### Which printers will be beneficial to this feature?

Marlin

### Describe the solution you'd like

Enable all filaments

Ergonomicmike, 2024-10-12:
Well, I have a lowly Ender 3 v2 and even tho TPU isn't listed as a filament (presumably "blacklisted" - or perhaps more properly, there simply isn't a profile for TPU) I print TPU anyway. I started with the standard Creality 0.4 mm nozzle Filament and Process settings and simply changed PA, Flow Rate, Retraction, speeds, etc. and made new User profiles.

Ergonomicmike, 2025-01-14:
Well if by "select TPU" filament you mean to have Orca actually recognize that you're printing with TPU, then you might be correct. (Although I just now opened "Filaments" in the latest dev Nightly and I see that there's a "My generic TPU" available. (But I didn't try it.))

Still, I don't understand why my workaround won't work for you. As above, I start with a PLA Filament preset, change the temps, disable retract, etc. and rename the Filament with TPU in its name. Same with Process. I start with PLA, slow the speed down, etc. and save the Process with TPU in its name. I've been printing TPU fine with Orca, even tho I haven't selected TPU filament proper.

madscientist42, 2025-01-15:
Why should someone HAVE to do a workaround on something like this?

Ergonomicmike, 2025-01-15:
Because someone decided that an Ender 3 v2 cannot print TPU.

timothynsheehan, 2025-02-22:
Having this issue as well with a direct drive upgraded Ender - there needs to be a way to enable this, Prusa slicer has no issue with it.

madscientist42, 2025-02-23:
> Because someone decided that an Ender 3 v2 cannot print TPU.

Heh...  They rather don't understand the crazies (*waves hand* Myself included) that do the, "impossible" on these cheapies.

I mean, who in their right mind tries to print Tullomer on even a 3V3 CoreXZ just because it's just within the envelope?

(Couldn't get layer adhesion right, so I'm moving to a Tungsten Carbide nozzle first and seeing what I can do to jimmy out 10 deg C out of the bed and raise the enclosure temps to 50-55 deg C.)

madscientist42, 2025-02-23:
To be sure, for the OrcaSlicer devs...you should really offer a bypass our barriers option.  Especially in the case of an Ender 3, there's a LOT of people that don't run stock HW in their printers.  We run amped up hot-ends, direct drive extruders, etc.

Technically, you can't print nylon or PC on the Stock V2...but before I burned out my Y-axis stepper pushing the poor thing to hard for it's own good, I'd been able to print almost anything shy of the Exotic filament types.  I'd already replaced the hotend with a RevoCR and was about to fix the TPU print problems with one of three possible mods for direct drive extrusion.