OrcaSlicer/OrcaSlicer #7088: Missing plus / minus for filaments

rashuge, 2024-10-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-rc

### Operating System (OS)

Windows

### OS Version

11

### Printer

QIDI Plus4

### How to reproduce

1. Open Orca Slicer
2. Switch to single filament machine
3. It displays two filaments available
[image]

### Actual results

No plus / minus to add and remove filament

### Expected results

Expect to have the ability to remove or add more filaments based on the number of filaments for that particular printer.  Or the ability to switch between the filaments.  The Qidi Plus4 when you select filament two, prints for a period of time then fails because it isn't filament one.  If I select filament one, it will complete the print without issues.  I would expect it to either use it like a filament profile and the printer completes the job or I would expect the ability to remove all but one filament since it is only single extruder, single filament.

### Project file & Debug log uploads

[V-Spooler X Beta 1.zip](https://github.com/user-attachments/files/17351617/V-Spooler.X.Beta.1.zip)
[debug_Sat_Oct_12_16_33_01_2272.log.zip](https://github.com/user-attachments/files/17351622/debug_Sat_Oct_12_16_33_01_2272.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

staal54a, 2024-10-13:
If you go into your Printer Profile Settings then click on the Multimaterial tab, is the "Single Extruder Multi Material" checkbox ticked? I was able to recreate this issue (on 2.2.0-beta2 so it's possible there's a new break between that and the rc) with the following steps:

1. Create a new Qidi X-Plus 4 profile (I don't own one so created a new one).
2. Open the Printer Profile settings, go to the Multimaterial tab and tick the "Single Extruder Multi Material" box
3. The filament plus/minus buttons appear so add a new filament.
4. Then, go back into the Printer Profile settings, Multimaterial tab and un-tick the "Single Extruder Multi Material" box

At this point you should have two filaments in the filament tab without the ability to remove the second one.

Using the steps you provided, I was not able to recreate the issue. Is it possible you maybe ticked the "Single Extruder Multi Material" checkbox for a single print, added the second filament then Orca closed or crashed? I just tried that as well and it did also recreate the issue you mentioned.

Archer36, 2025-01-03:
I a new user to OrcaSlier. I loaded a 3MF file into OrcaSlicer which I appears to have color information saved. I have a Voron 2.4 with a single extruder (not setup for multi material) and I too am missing the add and remove filament buttons.

Enabling multi material for my printer and restarting the slicer does not make the buttons appear. Any ideas?

wohltat, 2025-04-03:
I have the same issue.

[image]

"Single Extruder Multi Material" also solved it for me. I was pretty sure it was set before and i didn't turn it off.

closed 2025-07-10 by github-actions[bot]