OrcaSlicer/OrcaSlicer #7089: option to disable or tune UI smooth scrolling

bitingsock, 2024-10-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When scrolling through various UI elements it's apparent that the gui framework is using some sort of smooth scrolling feature. While this makes it easy to track moving elements it is done in a very poor way. While smooth scrolling to your line that you will end up at the GUI does not let you reverse directions or stop and will keep scrolling until it reaches the initial target. What's worse is that the speed is excessively slow (imo) and it buffers these movements so if you change directions multiple times rapidly it will slowly play through all the movements until the last one has been achieved. This is very jarring and feels like losing control of the GUI.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

An option to either disable smooth scrolling or tune it's speed would be sufficient but ideally the input lockout would be fixed as well.

### Describe alternatives you've considered

scrolling by grabbing the scroll bar does not use the smoothing feature, though there is still apparently an animation that slightly slows down scroll speed (this is not very obvious)

Dumplap, 2024-10-13:
yes

Felix14-v2, 2025-02-24:
I am infinitely surprised at how disgusting scrolling is done. It is insanely slow and is not smooth at all.

The best way to demonstrate the issue is the changelog window:

https://github.com/user-attachments/assets/1cccfb84-49c6-4a75-ab16-ee56d96fe5f8

That's on i5 12400H

Hamid-SHHP, 2025-07-11:
I want to confirm this issue – I’m experiencing the exact same problem with laggy and choppy mouse wheel scrolling in all OrcaSlicer menus and panels. It feels slow, laggy, and unresponsive, especially with fast direction changes or rapid scrolling.

**My system:**
- OrcaSlicer version: 2.3.0
- Windows 10 Pro, RTX 3080 Ti, Intel i7-4790K, 32GB RAM, ASUS VG279 (144 Hz, DisplayPort)
- Roccat Kain 122 AIMO mouse (tested with different polling rates and another mouse – no change)
- Mouse wheel scroll setting: standard (3 lines)
- No overlays, no exotic mouse software, everything up to date

Other Electron-based apps like Visual Studio Code scroll perfectly smooth on my setup, so this seems specific to OrcaSlicer.

I have tried all available workarounds (hardware acceleration flags, compatibility mode, etc.) – nothing improved it.

**Additional note:**  
I also notice a noticeable delay (about 1 second) when switching between settings tabs (e.g., Quality, Strength, Speed, Support, Other – in German: Qualität, Struktur, Geschwindigkeit, Stützen).  
The UI does not update instantly after clicking a tab.

This tab-switching lag is already described in issue #997:  
https://github.com/SoftFever/OrcaSlicer/issues/997

So it seems the overall UI performance (mouse wheel scrolling **and** tab switching) could be improved.

Thanks for looking into this!

Felix14-v2, 2025-10-04:
> UI performance issue seems related to specific Windows 10 setups (smooth on Windows 11 & Linux)

I can confirm that this behavior also affects Windows 11