OrcaSlicer/OrcaSlicer #7092: Import Cura Conical Support

guigallina92, 2024-10-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

no

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Import conical support from cura
https://www.youtube.com/watch?v=32t9wUYO8mg

guigallina92, 2024-10-14:
Please import this feature, it helps a lot to reduce time and material expenditure on supports.

Noisyfox, 2024-10-21:
That's a very interesting idea and makes a lot of sence. Will see if we could bring it to Orca.

guigallina92, 2024-10-21:
> That's a very interesting idea and makes a lot of sence. Will see if we could bring it to Orca.

Thank you, it will be of great use

closed 2025-01-27 by github-actions[bot]