OrcaSlicer/OrcaSlicer #7093: Filament Setting Overrides don't work for Bambu A1 Mini

S1lentA0, 2024-10-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Bambu A1 Mini

### How to reproduce

-Go to your list of filament and 'Click to edit preset'
-Go to Setting Overrides tab
-In my case change Retraction lenght, Retract on layer change, Retraction speed, Wipe while retracting.
-Slice print
-In preview retractions are still applied at the layer change.

### Actual results

When I apply these settings to my TPU filament profile, I see these changes are not applied to the preview and during the print. The extruder clearly retracts every layer change when printing for example a 1 wall 0 top layer 0% infill test cube. 

I tried disabling retraction in the A1 Mini machine Gcode to see if this causes some kind of override, but when disabling retraction on layer change there, it  still doesnt apply the setting to the print.

When switching to my P1S on the same print and settings, this issue doesn't occur. 

### Expected results

Expected result is quite obvious, when disabling filament retraction on layer change, the slicer should not add a filament retraction to the gcode. 

### Project file & Debug log uploads

[Cube.zip](https://github.com/user-attachments/files/17354808/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]
Retraction shows up in preview screen and gcode

[image]
When applying same same settings to P1S printer, retraction is disabbled accordingly

tiziano149, 2024-12-10:
#7719 Same issue

S1lentA0, 2025-05-04:
Bumping this issue. Issue still persistant in the current stable version. Just checked with the most recent build of BL and seems the issue has been fixed in the Bambu Slicer.