OrcaSlicer/OrcaSlicer #7094: Can't get rid of Solid Infill

otede, 2024-10-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

 Intel i3-12100F

RTX 2070 8GB MSI

64 GB RAM

### Printer

Bambu Lab A1

### How to reproduce

1. Slice a model where you'd like very thin horizontal shells and the rest with infill
2. Disable Ensure Vertical Shell Thickness

### Actual results

Lot's of internal solid infill! In many cases up to 40% material wasted.

### Expected results

Close to zero solid infill.

### Project file & Debug log uploads

[log 2024-10-13.zip](https://github.com/user-attachments/files/17354940/log.2024-10-13.zip)
[converter voltage QSKJ 1853869 step buck projektorca 02.zip](https://github.com/user-attachments/files/17354941/converter.voltage.QSKJ.1853869.step.buck.projektorca.02.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

igiannakas, 2024-10-15:
Reduce your top shell thickness to 0mm (not just the number of layers but also the amount in MM that it is extruded).

otede, 2024-10-16:
Thanks for chiming in @igiannakas this is pretty much it. BUT. It's not a fix or a misuse on my part.
Let's recap:
EVST is designed to prevent [~0 top shell thickness]
EVST either goes overboard with the amount of material OR refuses to be disabled when EVST is set to 'None'
EVST only considers itself disabled when [~0 top shell thickness] is specifically set by the user.

To put it into perspective it's as if Ironing featured refused to disable unless the user specifically state that he wants the top surface in the poorest condition possible. 

Notably, in case of that enclosure module included in the attached project it basically "solves" the problem, in other models, more akin to figurines, setting [~0 top shell thickness] causes gaping holes to occur in the print.

igiannakas, 2024-10-16:
Ensure vertical shell thickness looks at slanting walls, not top surfaces. Solid infill is generated on both walls and surfaces with ensure vertical shell thickness enabled, to ensure the wall is of a consistent thickness when sloped. 

The options reduce the sensitivity of the algorithm for sloped walls but do not affect flat top surfaces, which continue to generate solid infill as per the mm and number of layers settings.

have you tried my suggestion above? What did it do with 0mm thickness?

in general EVST to none is not recommended for any models that you care about structurally by the way. It’s use is or should be limited to speed runs and specific exceptions.

otede, 2024-10-16:
@igiannakas Yes, I have tried it and it actually got rid of the excessive solid infill layers. I just went on with explaining that this cannot possibly be considered 'OK', even in UX-nightmares Prusa-derivatives are.

igiannakas, 2024-10-16:
That is why it’s called **vertical** shell thickness and not horizontal :) it governs the consistency of the thickness of the walls.

the thickness of the top and bottom surfaces is governed by the top and bottom shell thicknesses :)

otede, 2024-10-16:
First, the 'vertical' refers to the directionality of the cross section plane, not direction of shells. Second, the very point of this bug report is that that EVST interferes with the Orca's controls of horizontal feature thickness.

closed 2025-01-23 by github-actions[bot]