OrcaSlicer/OrcaSlicer #7096: Adding objects/cloning objects to second plate is auto arranging all plates together and deleting second plate

tastyratz, 2024-10-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2 beta

### Operating System (OS)

Windows

### OS Version

10

### Printer

Voron Trident

### How to reproduce

Create second plate with parts, drag STL's onto plate, clone the objects to increased qty

### Actual results

While trying to create 2 plates for 2 different prints I am dragging a part onto the window/plate2 and then cloning that part to the qty needed. Around half of the time when doing so the clones are created, auto arranged, and just fine on plate 2. The rest of the time, the parts are all removed from plate 2, arranged on plate 1 with my original plate layout, and then removing plate 2 altogether.

### Expected results

Parts including their clones should honor the original plate placement.

In addition, cloning a part changes my custom plate arrangement now. Would it be possible to auto arrange the new clones onto an existing build plate INSTEAD of rearranging the entire build plate because I cloned an object?
That's especially important for people with gouged plates trying to avoid known scars
[image]
[image]
[image]

### Project file & Debug log uploads

Plate 1 had parts for latest XOL revision
https://github.com/Armchair-Heavy-Industries/Xol-Toolhead/blob/main/docs/printing.md

Plate 2 had parts from AWD support parts here
https://github.com/aTinyShellScript/v2.4_AWD/tree/main/STLs/NEMA17/MetalPlates

### Checklist of files to include

- [ ] Log file
- [ ] Project file

elcuro, 2024-10-18:
I have the same issue with Bambulab X1C
 
Before:
[image]

After:
[image]

tastyratz, 2024-10-18:
> I have the same issue with Bambulab X1C. It removes not only second plate, but all others.
> 

Is this upstream in Bambuslicer for you as well or were you saying this is happening on your X1C with Orca?

elcuro, 2024-10-18:
It happend it to me on Orca 2.2.0-beta on Windows 11 Pro to be clear.

Noisyfox, 2024-10-19:
It's been fixed in beta 2.

jannoke, 2024-10-20:
Had the same problem. skipped beta2 and just installed 2.2.0RC and it works now

closed 2025-01-27 by github-actions[bot]