OrcaSlicer/OrcaSlicer #7099: Crashing on Linux. Again. Old fix no longer works.

trebory6, 2024-10-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 Beta

### Operating System (OS)

Linux

### OS Version

Nobara Linux 40 (KDE Plasma) x86_64

### Additional system information

Kernel: 6.11.3-200.fsync.fc40.x86_64 
Uptime: 4 hours, 23 mins 
Packages: 3286 (rpm), 55 (flatpak) 
Shell: bash 5.2.26 
Resolution: 2560x1440 
DE: Plasma 6.2.0 
WM: kwin 
CPU: AMD Ryzen 5 5600X (12) @ 4.651GHz 
GPU: NVIDIA GeForce RTX 4070 SUPER 
Memory:  32009MiB 

### Printer

Elegoo Neptune 3 Pro

### How to reproduce

Trying to open v2.2.0 Beta on Linux after recent package updates results in immediate crash of Orca Slicer. 

### Actual results

Trying to open v2.2.0 Beta on Linux after recent package updates results in immediate crash of Orca Slicer. 

### Expected results

Open and use Orca Slicer without issue. 

### Project file & Debug log uploads

[Ultimate Drybox.zip](https://github.com/user-attachments/files/17357357/Ultimate.Drybox.zip)

[debug_Sun_Oct_13_15_.zip](https://github.com/user-attachments/files/17357370/debug_Sun_Oct_13_15_.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

So originally the steps in [this reddit post](https://old.reddit.com/r/NobaraProject/comments/1etefw2/this_might_be_a_long_shot_but_has_anyone_gotten/) worked to fix the issue. At some point recently it no long works and now crashes as it did before. 

Log File consists of the following: 

[error]	2024-10-13 15:37:06.440864[Thread 0x00007ffb62ef1700]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2024-10-13 15:37:06.686551[Thread 0x00007ffb62ef1700]:03:37:06 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-10-13 15:37:06.692066[Thread 0x00007ffb62ef1700]:03:37:06 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-10-13 15:37:06.692606[Thread 0x00007ffb62ef1700]:03:37:06 PM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-10-13 15:37:06.734383[Thread 0x00007ffb62ef1700]:03:37:06 PM: Warning: Error running JavaScript: Unsupported result type
[error]	2024-10-13 15:37:08.130639[Thread 0x00007ffb62ef1700]:calc_exclude_triangles:Unable to create exclude triangles

sharkovich, 2024-10-14:
For me OrcaSlicer also stopped working today. Opens up, worsk for like 5seconds then hangs and closes. Console says segmentation fault. Worked fine just few hours earlier till the system auto update kicked in. 2.2.0-rc from AppImage. Did a rollback and we are back to normal. Please let me know what could I provide.

```
Operating System: Bazzite 40
KDE Plasma Version: 6.2.0
KDE Frameworks Version: 6.7.0
Qt Version: 6.7.2
Kernel Version: 6.9.12-210.fsync.fc40.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 5800X 8-Core Processor
Memory: 15.5 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 3080/PCIe/SSE2
Manufacturer: ASUS
```

CitrusIntellect, 2024-10-15:
Same issue on Arch Linux, I am 99% sure an nvidia GPU causes the issue as on my other device with an AMD GPU, there are no issues

trebory6, 2024-10-15:
> Same issue on Arch Linux, I am 99% sure an nvidia GPU causes the issue as on my other device with an AMD GPU, there are no issues

Yeah it's something to do with the Nvidia GPU and the Webkit that Orca uses, since the previous fix had you installing a specific webkit and forcing Orca to use it. Super frustrating as it has been an issue for a while now(#5171, #1841, #3479) with no movement on a fix.

trebory6, 2024-10-15:
I will add that downgrading to 2.1.1 seems to still work with the linked WebKit fix.

dkowis, 2024-10-26:
I am using the 2.1.1 official version, and I have to do the linked Webkit fix:
```
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./OrcaSlicer_Linux_V2.1.1.AppImage
```

I'm on Pop!_OS 22.04 LTS x86_64
Kernel: 6.9.3-76060903-generic
CPU: AMD Ryzen 9 3950X (32) @ 3.500GHz
GPU: NVIDIA GeForce RTX 3080 Ti

I think it started happening with the latest version of the nvidia drivers: nvidia-driver-560

dkowis, 2024-11-02:
Still happens on 2.2.2, but the EGL fix is usable.

trebory6, 2024-11-18:
This was working for a while, but for some reason does not work anymore. 

Here is the errors I get when I run it from the terminal:

```
/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
[2024-11-17 19:40:53.648131] [0x00007fb354a867c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.305: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.305: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:33.305: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.305: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.306: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.306: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:33.306: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.306: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:33.307: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.307: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:33.307: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.307: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:33.308: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.308: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:33.309: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.309: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:33.310: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.310: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:33.310: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.310: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.311: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.311: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:33.311: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.311: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.312: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.312: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:33.312: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.312: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:33.346: gtk_window_resize: assertion 'height > 0' failed

[... 283 lines cut ...]
(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:35.966: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:35.966: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:35.966: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:35.966: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:35.968: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:35.968: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:35.970: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:35.970: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:35.970: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:35.970: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:31393): GLib-GObject-CRITICAL **: 19:41:35.970: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:31393): Gtk-CRITICAL **: 19:41:35.970: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

```

mstraciak, 2024-11-21:
i have the same issue. i have mxlinux (debian 11)
for me works uninstall libwebkit2gtk-4.0-37 and libjavascriptcoregtk-4.0-18 packages.
manualy install these packages:
libjavascriptcoregtk-4.0-18_2.44.2-1~deb11u1_amd64.deb
libwebkit2gtk-4.0-37_2.44.2-1~deb11u1_amd64.deb

probably package libwebkit2gtk-4.0-37_2.46.3 is incompatible with orcaslicer appimage.

anarsoul, 2024-11-22:
Please try if flatpak from https://github.com/SoftFever/OrcaSlicer/issues/3949#issuecomment-2483844840 fixes the issue

aaravrav, 2024-11-22:
@anarsoul It does not.

aaravrav, 2024-11-22:
> Still happens on 2.2.2, but the EGL fix is usable.

I can't seem to find the "2.2.2" version, although I did try this with 2.2.0 and 2.1.1, with all of them crashing instantly or after clicking a few buttons. Where is the 2.2.2 version?

dkowis, 2024-11-22:
> > Still happens on 2.2.2, but the EGL fix is usable.
> 
> I can't seem to find the "2.2.2" version, although I did try this with 2.2.0 and 2.1.1, with all of them crashing instantly or after clicking a few buttons. Where is the 2.2.2 version?
> 

https://github.com/SoftFever/OrcaSlicer/releases/tag/v2.2.0

aaravrav, 2024-11-23:
That's 2.2.0, not 2.2.2

trebory6, 2024-11-23:
So this is a bit of a workaround but I just got a Saturn 4 Ultra and I needed to print some stuff for an HVAC/Ventilation setup so I can start printing ASAP and I got absolutely fed up with the constant issues that OrcaSlicer has on anything other than Ubuntu and brute forced my way into finding a way to get Orcaslicer to run reliably on my Fedora desktop. 

Theoretically this should work on any distro, including incompatible versions of Ubuntu, since it uses Distrobox to create a Ubuntu 22.04 container to create a terminal environment with known compatibility and known working dependencies and runs a compiled version from there.

With Environment Variables I was able to get OrcaSlicer 60% working with occasional crashes and with some things that just didn't work. The way I've now done it has no crashes(so far) and everything works the way it's supposed to. ONLY slight issues is that I've noticed that viewing things in 3D isn't as smooth, so if you move it around it feels like low FPS. But I'll take it.

I wrote a post on Reddit [here](https://old.reddit.com/r/OrcaSlicer/comments/1gxuiu4/linux_tutorial_on_how_to_get_a_reliable_version/) explaining how to do it. Forgive the crudeness, I was working on a lack of sleep, and I'm mostly self taught so admittedly I'm not an expert.

aaravrav, 2024-11-23:
@trebory6 Following your procedure, running `./BuildLinux.sh -dsir ` gets ```
[10/130] Creating directories for 'dep_OpenSSL'
ninja: build stopped: subcommand failed.
```

trebory6, 2024-11-24:
> @trebory6 Following your procedure, running `./BuildLinux.sh -dsir` gets
> 
> ```
> [1/130] Performing patch step for 'dep_OpenEXR'
> FAILED: dep_OpenEXR-prefix/src/dep_OpenEXR-stamp/dep_OpenEXR-patch /home/aarav/Downloads/OrcaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR-stamp/dep_OpenEXR-patch 
> cd /home/aarav/Downloads/OrcaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR && /usr/bin/git apply --verbose --ignore-space-change --whitespace=fix /home/aarav/Downloads/OrcaSlicer/deps/OpenEXR/0001-OpenEXR-GCC13.patch && /usr/bin/cmake -E touch /home/aarav/Downloads/OrcaSlicer/deps/build/dep_OpenEXR-prefix/src/dep_OpenEXR-stamp/dep_OpenEXR-patch
> fatal: unknown style 'zdiff3' given for 'merge.conflictstyle'
> ```
> 
> ...
> 
> ```
> [10/130] Creating directories for 'dep_OpenSSL'
> ninja: build stopped: subcommand failed.
> ```

The error fatal: unknown style 'zdiff3' given for 'merge.conflictstyle' typically occurs when the merge.conflictstyle=zdiff3 configuration option is set in Git, but the version of Git being used does not support the zdiff3 merge style. The zdiff3 merge style was introduced in Git 2.39, so using an older version of Git (pre-2.39) can lead to this error.

If all else fails, you can delete the distrobox, delete the git clone, and try again. That's the good thing about Distrobox is that it's contained, so every time it's a fresh slate. 

Make sure you're running git version 2.34.1 by running this command:

`git --version`

D0m-Sch, 2024-11-26:
It doesn't work on Ubuntu-Studio 24.04 either. Same crash even with the special Ubuntu version of the .AppImage.

trebory6, 2024-11-26:
> It doesn't work on Ubuntu-Studio 24.04 either. Same crash even with the special Ubuntu version of the .AppImage.

I will say that the tutorial I linked is confirmed, at least on my own personal end, to work with the Ubuntu 22.04 Distrobox terminal environment. Haven't tested it on anything other than my Linux system, but I can't see why it wouldn't work for other distros.

D0m-Sch, 2024-11-26:
I will give it at try when I have time.
But having to find a workaround misses a bit the point, doesn't it.
Especially since there is a specific Ubuntu2404 version of the software.
And who knows how long the current workaround will work. 
I personally have a separate SSD with Win10 and it works just fine with that setup. But, I wouldn't call that an acceptable solution.

trebory6, 2024-11-26:
I mean, theoretically the workaround I linked to should work indefinitely since it's technically Ubuntu 22.04 being run inside a terminal container. You can update your OS all you want, but the container will remain version 22.04.

And trust, me I know. Orcaslicer's linux support definitely needs improvement, but I am of the "Fine, I'll do it myself" mentality.

n00b0ss, 2024-11-26:
https://myside-yourside.net/@StarkRG/113526784879479099

aaravrav, 2024-12-01:
> The error fatal: unknown style 'zdiff3' given for 'merge.conflictstyle' typically occurs when the merge.conflictstyle=zdiff3 configuration option is set in Git, but the version of Git being used does not support the zdiff3 merge style. The zdiff3 merge style was introduced in Git 2.39, so using an older version of Git (pre-2.39) can lead to this error.
> 
> If all else fails, you can delete the distrobox, delete the git clone, and try again. That's the good thing about Distrobox is that it's contained, so every time it's a fresh slate.
> 
> I'm running git version 2.34.1 within the Ubuntu 22.04 container. Make sure you're running the right version by running this command while inside your Ubuntu distrobox:
> 
> `git --version`

Nope, it does not work on the version. You need to update git first, using the following commands:
```bash
sudo apt-get install software-properties-common 
sudo add-apt-repository ppa:git-core/ppa -y
sudo apt-get update # this will update git if you already have it. Install git if you don't already
```

aaravrav, 2024-12-01:
Can confirm it works, but it is terribly slow. This is because distrobox by default does not have access to your GPU. Now for me, if I try to add my gpu to distrobox (the way it is supposed to be added), it fails to do so. Even if it could, I fear that due to nvidia orca-slicer would crash again

stfngnsch, 2024-12-06:
My problems are only related to network issues. Without adding the Hostname or IP for my physical printer, the slicer does not crash. When I add a remote connection, it crashes within 5 seconds.

n00b0ss, 2024-12-06:
@stfngnsch try add a port to the hostname or IP

stfngnsch, 2024-12-06:
Thanks you very much, this kinda works. 7125 is the port for moonraker. Now I can slice and upload. But I can still not access to moonrakers front end. 

This solution was already presented here:

https://github.com/SoftFever/OrcaSlicer/issues/7210#issuecomment-2440350811

stfngnsch, 2024-12-06:
> @stfngnsch try add a port to the hostname or IP

Thanks you very much, this kinda works. 7125 is the port for moonraker. Now I can slice and upload. But I can still not access to moonrakers front end.

This solution was already presented here:

https://github.com/SoftFever/OrcaSlicer/issues/7210#issuecomment-2440350811

closed 2025-03-15 by github-actions[bot]

ramonskie, 2025-09-22:
im currently experiencing this with the latest fedora and updates
i have a amd cpu with amd igpu and nvidia dgpu
fedora is using hybrid graphics.
im not sure where to go from here