OrcaSlicer/OrcaSlicer #7100: Add setting to toggle filament start index with one(default) or zero(new)

BrunoSouzaPicinini, 2024-10-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In MMU setup the gcode used to change tools starts with zero index (`T0`).
To avoid confusion add a toggle in settings where user can set presentation to start with zero instead of the current one.

Also this change will affect the keyboard numbers shortcut.

Eg.: Print of section that will be modified 
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

When option is disabled the current behavior is maintened.
When option is enabled the first filament number shown is zero and keyboard shortcut is `0` instead of `1`.

### Describe alternatives you've considered

The MMU could show numbers starting from one but the gcode itself still would be `T0` with is counter intuitive.

lorangthierry, 2024-11-13:
Great Idea. Mapping Tool to Gate would be much easier

sorted01, 2024-12-20:
I am struggling with that since i started with MMUs

meganssmith, 2024-12-29:
Yes please

kens96, 2025-03-02:
Yes please!

JonasBouchard, 2025-04-02:
Yes please!

gonchogsm, 2025-04-23:
Great idea! Up!

kens96, 2025-04-23:
Unfortunately this has been open for over 6 months but doesn't seem to have any traction, with HH and MMUs becoming more common place it is a great enhancement
++1

SHir76, 2025-06-13:
That would be really great. As far as i know, most klipper based printers count from zero.
I.e. i have a tool changer and have already several times  heated up the wrong extruder by accident to change filament because it is extruder 2 for filament 3 etc.

kens96, 2025-11-23:
> Hey everybody! I am currently working on that and I think it will work. It is on my Github right now, but I want to test it before doing a pull request. Can anybody help me to build my own OrcaSlicer branch? I have tried on Windows and Linux and I continuously get errors while compiling. Any help would be greatly appreciated!

Hi Jonas, have you been able to make any progress with this? I think it would be a great addition.