OrcaSlicer/OrcaSlicer #7106: Allow choosing different filament for outer walls, inner walls and infill.

Festivejelly, 2024-10-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Printing with matte filaments look great, but they are not strong. How about being able to print infill and inner walls with a strong material such as PLA+ then printing the outer walls with a different filament.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Be able to choose the filament for infill, inner walls and outer walls.

### Describe alternatives you've considered

This can be worked around by creating a shell for your model and then importing them both, but its time consuming and doesnt work too well.

53Aries, 2024-10-15:
I support this. SEMM is no different than multiple extruders in this regard. We just need a T{x} command for the line type being printed. 
[image]
[image]
OrcaSlicer does have this functionality though only for multiple extruder systems, and it doesn't seem to be functional. When SEMM is selected the option to specify filament for features is disabled.  
[image]
[image]
[image]

BigD1950, 2024-11-19:
I would like to print the outer wall(s) with a CF reinforced filament and the inner structure with a different filament.
I am developing a pellet feed hotend to use recycled PET, this would be adequate and strong enough for most projects. 
Printing the outer wall with the desired substrate would complete the process?

bo0tzz, 2025-02-18:
No it's not

Festivejelly, 2025-02-18:
Not stale it just needs to be implemented the feature request has enough detail on it.

KoenVanduffel, 2025-02-26:
I second this request.
I find that there is the "Multimaterial - Filament for features" section where different filaments can be selected for walls, infill, ..., however if I select the second filament for infill it keeps using the first filament for the whole print. I have an IDEX printer so this function simply seems not to work, or am I missing something?

brettvitaz, 2025-03-03:
I look forward to this feature! I have been making great use of it on the XL and PrusaSlicer by using less expensive / surplus filament for infill.

BigD1950, 2025-03-03:
This is my plan to a T, I am aiming for pellet feed hotends, if we can have
outer wall in separate color, virgin polymer; then all internals can be
printed from recycled filament (Yucky Grey, seems to be common), I think
more will go for recycling if we can choose what is printed from where at
the slicer.

KoenVanduffel, 2025-03-03:
LOL, i work in the PVC industry where it is already common practice to use recycled material for the core and only the outer wall in virgin material.

I was thinking of outside 1, 2 or 3 walls in TPU and inside is PETG or ABS to get very tough and stiff parts (at least a better balance between both). I do that now in CAD by giving the model an outer shell body. This is very intensive to design and can be poorly tweaked. Just being able to specify x outer layers with a different extruder (material) would make this mainstream very easily.

fanjules, 2025-04-18:
I would love to see this feature too.  With the Bambu H2D, it would be a powerful combination (and indeed the Prusa XL, or more wasteful material changers if you don't mind the poop).

I'm going to go as far to say it's a game changer in terms of creating composite material physical or aesthetic properties.

bo0tzz, 2025-04-19:
Folks there's an upvote button on the original post, please use that instead of sending unnecessary notifications to everyone who is subscribed here.

fanjules, 2025-04-19:
It's not an unnecessary notification when I hear another person has joined our growing army... 🤣

fanjules, 2025-12-04:
A simple implementation of this was added in Bambu Studio 2.04.  They reference parallel implementations in Prusaslicer and Orcaslicer, but I think those parallel implementations refer to different toolheads?  In Bambu, you can do it for a simple AMS and single nozzle, even on an A1 mini.  

[image]

The settings available are quite simple, you choose filament for sparse/solid infill and walls.  One day I hope we can have just the outer wall in one particular filament.

The feature is only available if you enable developer mode in settings.

[image]

spartipilo, 2025-12-07:
Hi, On OrcaSlicer with Snapmaker U1, when I select to change filament for infill, doesn't work. 😔 Solutions?

luderunner, 2026-01-08:
> A simple implementation of this was added in Bambu Studio 2.04. They reference parallel implementations in Prusaslicer and Orcaslicer, but I think those parallel implementations refer to different toolheads? In Bambu, you can do it for a simple AMS and single nozzle, even on an A1 mini.
> 
> <img alt="Image" width="1114" height="477" src="https://private-user-images.githubusercontent.com/10921550/522415145-a2f7238c-6d25-49c5-b5c7-281cb68fb4e3.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njc4ODI1NjAsIm5iZiI6MTc2Nzg4MjI2MCwicGF0aCI6Ii8xMDkyMTU1MC81MjI0MTUxNDUtYTJmNzIzOGMtNmQyNS00OWM1LWI1YzctMjgxY2I2OGZiNGUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMTA4VDE0MjQyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJkYjcxZjgxMWVhMzg3OTU0Yzg3Y2FmMjg1MGRhZTQ5N2Y1OGIwOTY1MDkzZDdjYWI5YzA1ZmY5NmRmMjBiYTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.mFgRTSK4J0oVMUcPwAiITkVeQqujo_b19i_eav__DFc">
> The settings available are quite simple, you choose filament for sparse/solid infill and walls. One day I hope we can have just the outer wall in one particular filament.
> 
> The feature is only available if you enable developer mode in settings.
> 
> <img alt="Image" width="389" height="214" src="https://private-user-images.githubusercontent.com/10921550/522416390-443a68ef-263b-4445-91fb-4e3e2e60350d.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njc4ODI1NjAsIm5iZiI6MTc2Nzg4MjI2MCwicGF0aCI6Ii8xMDkyMTU1MC81MjI0MTYzOTAtNDQzYTY4ZWYtMjYzYi00NDQ1LTkxZmItNGUzZTJlNjAzNTBkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAxMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMTA4VDE0MjQyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg1MDliNTliN2Q5ODNkYjA2ODUxNmQwNzgzNjhjYmRkYTFmMjg5MThmYTlhM2JlODM5NmU1MTY3ZGJjZTY5Y2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ly5PeqptLOo6i9TN2rL3JwZ12FDHw4R0d9PAW9oNBKo">

I can't seem to get these options in either Orca or Bambu, even with develop mode turned on..