OrcaSlicer/OrcaSlicer #7107: 3D Navigation is difficult, orientation is hit or miss.

sonjamichelle, 2024-10-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Navigating the 3D workspace in Orca can be quite challenging. The view feels uncoordinated and difficult to control, unlike other 3D applications I've used.

This problem occurs in both Perspective and Orthographic views.

In contrast, programs like 3DS, AutoCAD, Revit, Blender, and BambuStudio make it easy to manipulate the view, allowing for precise work. However, Orca is more like driving a car without brakes or steering - it takes me a lot of time to get the view at the right angle to see what I'm working on. Typically, I need to use the view cube to regain control of the view before retrying to orient it.

As someone who has been using 3D applications since 2003 (starting with GMax), I'm not new to navigating views. In fact, I don't experience this issue with BambuStudio or AutoDesk's products or SolidWorks and FreeCAD to name a few. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Finer control without resorting to using the 3D Cube.

### Describe alternatives you've considered

Not using the application.

MxBrnr, 2024-10-15:
I would argue the exact opposite. Orca's 3D navigation feels FAR more intuitive for basic functions than those other programs. Left-click to rotate, right-click to slide, scroll to zoom. Easy, and it is the same as many other 3D software use, like Google Earth, Windows 3D programs, and most online 3D viewers (such as GitHub, Thingiverse, and Printables). I wish those other CAD programs would revise their navigation because they are really over-complicated, especially for beginners, or at the very least, give the option of choosing what kind of control you want. I think Autodesk Fusion is the worst about this; their controls are horrendous. It seems to me that Orca is not aimed at CAD pros, so I don't expect that the controls here would match elsewhere (and am personally glad they don't).

foreachthing, 2024-10-15:
Left click to rotate is just a mess.... I fully agree with the rest wirh @sonjamichelle .
One always has to be sure not to click on anything. 
I'm glad I got a 3Dx device. So, every application zooms and pans the same.

sonjamichelle, 2024-10-15:
It is NOT whether the interface is "intuitive" or not.

It comes down to the fact that when you are ROTATING the view if feels wildly "out of control".

With BambuStudio, as you rotate the view fluidly rotates around what you have centered. The movements are fine and controlled. Same with pretty much every other 3D app I have installed, regardless of whether they are "intuitive" or even have the same buttons, keys assigned to function.

I AM NOT SAYING THE WAY ORCA DOES IT IS "NOT INTUITIVE"!!!

I am saying that the "speed" we'll call it is too much, you have no fine movements over the rotation or orientation.

since most likely folks are doing the TLDR and not actually READING what I am saying, here are two clips demonstrating the point.

https://github.com/user-attachments/assets/fe0dfb8a-141b-4c51-9d21-599b3b6933bd

https://github.com/user-attachments/assets/e5877ea2-781d-47c7-b692-ddea82b7733f

The key bindings to rotate are the same, who cares at this point. IT IS THE ACTUALL MOVEMENT that is not correct, out of control in Orca.

You can see that in Bambu the movement is smoother, easier to focus and work on the items where Orca is like it is speeding along and you spend more time just trying to orient the view to get into what you want to work on.

tlhintoq, 2024-10-16:
I'll throw my hat (and comments) in the ring with @foreachthing but be a little more direct about it...

Anyone that has been doing 3d work since 2003 and with a litany of 3d program experience under their belt, should consider spending a few bucks on a 3d mouse and not have to concern themselves with different competing ways of representing 3d space with a 2d  mouse.
Generally you can get a Spacemouse Pro for about 100 bucks, less if you keep your eyes peeled for a couple weeks.
[image]

But to be fair, I don't disagree about rotating the work with the same left-click and object selection: That's just nutz.
Its one reason I suggested an option to simply not navigate with the 2d mouse if there is a 3d mouse connected: there's no reason to. https://github.com/SoftFever/OrcaSlicer/issues/6598

tlhintoq, 2024-10-16:
> The concept of, _"Well they won't fix it so here's what you'll have to do."_ is just pure BS and laziness on the devs part.
Well... I didn't say that.  I don't see anyone that did.  I mentioned to you that it sounded like you'd benefit from it.  You brought up your years of 3d and range of software so I don't think it was such a far fetched suggestion for circumstances you choose to share as part of justifying the changes.

> I'm going to hazard a guess that the MAJORITY of Orca users are NOT 3D designers,
Again - I wasn't talking to the majority of Orca users.  I was talking to you, within the context you brought in to the conversation.  Please don't try to recontectualize my comments aimed at helping you specifically to make it sound like I'm suggesting everyone everywhere get a 3d mouse because Orca developers are lazy.  I never said any such nonsense.

I also agreed with you that the current setup is, and I quote, "nutz".  So maybe a full screen of angry rebuttal to someone that agreed with you and tried to offer a helpful suggestion to you specifically (not the whole world) in the context you started, is maybe a bit overboard.

As for "maybe it would be nice" - well, as a 60 year old in the same space I can tell you that it took me about a day to adapt to it and LOVE it.  I'll never give it up.  Most all my software will use it except for Windows 3d builder and that sucks.
I started with a used $45 Spacemouse Compact and in a week regretted not getting the Pro with all the modifier and function buttons to reduce the travel from Spacemouse to keyboard.  Now I can do 90% of my 3d work without having to move my hands from the two peripherals.
[image]

sonjamichelle, 2024-10-25:
One thing I just noticed.

When I unchecked "Use Free Camera" it "seems" like it is easier to navigate.

Not a 100% sure. I did switch my desktop to Linux (PopOS with KDE). At first it was still "wild" then I got the odd idea of unchecking that box. We'll see.

sonjamichelle, 2024-10-26:
This held true on my Windows 11 VM. Unchecked the "Use Free Camera" and the view is easier to manage. Not perfect, but a LOT better than with it checked.

Not sure if this is check by default or not.

silvernull, 2025-01-24:
It would be nice if camera rotated *around whatever point the cursor is on* also instead of the centre of the selected object for easier navigation.

GabrieleCzz, 2025-02-17:
I also would like to have the camera rotating around the object like in most of 3D tools. Orcaslicer is doing like that selecting the object in "prepare", but in "preview" it is rotating around the center of the plate that is very inconvenient.

Felix14-v2, 2026-10-07:
Could you check the solution I posted in https://github.com/OrcaSlicer/OrcaSlicer/issues/5026#issuecomment-6029186092? I think it covers almost all problems mentioned here.