OrcaSlicer/OrcaSlicer #7109: Filament Swap Reduction via Auto-Rotation

v2thegreat, 2024-10-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When I was on Reddit, I found [this post](https://www.reddit.com/r/BambuLab/comments/1g2qk29/what_is_a_normal_amount_of_waste_for_multi_color/) on filament waste, and it got me thinking: If the print was rotated differently, then the number of filament swaps could have been reduced, saving a ton of purging and time.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Auto rotating the model in such a way that it can find the best position to minimize filament swaps.

### Describe alternatives you've considered

1. Rotating it myself 

### Additional context

I made a 2D prototype in Python to see if this idea has any merit. You can find the full notebook [here](https://gist.github.com/v2thegreat/a5dbf01e6b582fb66091db47bc05e6cb).

The idea that I had to test this out was given an image/model/projection, I want to find the best rotation (orientation) that minimizes the number of colors in a layer, minimizing the filament swaps that need to happen. 

For my prototype, I took a brute-force approach simply rotating the image until I found one that minimized the number of swaps. You could further optimized this by using a solver or some other form of simulated annealing or just some better math since in 3D space, we'd be dealing with 90 * 90 = 8100 degrees that would need to be tested. This might be overengineering on my part since I'm using python and most slicer software seems to be made using C++.

I used a scoring function as the average number of colors across all layers, and tried to minimize that.

Based on my testing, I think this appraoch can work quite nicely. Here are some graphs to back me up:

### Image with random spots of different colors

[image]

### Random Swipes
[image]

### Stich (since he started it all)
[image]

Sorry for my messy work, it's thanksgiving here in Canada and I have a few things to take care of. You can find my exploratory notebook [here](https://gist.github.com/v2thegreat/a5dbf01e6b582fb66091db47bc05e6cb)!

I would've done this in 3D, but I couldn't find a way to extract color information from the 3MFs I got from makerworld (since that's the main usecase). If someone knows how to access 3D models with color so I can continue to prototype (preferably in Python), please let me know! I'm also not very familiar with slicer programming in general, so I might've made assumptions here that were wrong.

igiannakas, 2024-10-15:
This is certainly interesting! However how do you deal with the reality that most models can only really be oriented in a handful of ways to optimise for 3D printing, due to the presence of overhangs and needed supports? For example in your picture above, the optimised object, while having less filament swaps, it also is pretty much unprintable due to the almost non existent contact patch with the bed.

v2thegreat, 2024-10-15:
As with all engineering, there are probably tradeoffs to different approaches. Given your experience, let me know what makes the most sense. I'm just proposing these off the cuff.

1. Adjust the score function to include a stability score, and the user can decide how each score is weighted against the other. For example, the user can maximize stability while minimizing print swaps (1:1), maximize stability while ignoring print swaps (1:0), ignore print stability and minimize print swaps (0:1) based on their preferences.
   1. The stability score could be derived from the contact area with the first layer, combined with a measure of overhangs and such. However, this does introduce technical complexity.
2. In the UI, you could orient the part but warn that this approach would be optimal for colour swapping but might not be for stability and surfaces. It might even make sense to propose that the user prints the model first in a single colour to validate that it is stable enough to succeed (since single-colour printing is much faster than multi-colour printing), then print in multicolour. 
   1. This does introduce additional costs, but if you're printing a plateful or multiple platefuls of parts, I'd argue that the tradeoff is worth it since the time saved will be spread across all the parts you're printing and should outweigh the initial print pretty quickly.
3. Adjust the model to cut off a small piece to increase surface area, then print both parts and glue them together. This is something that I've done before, and even a tiny chamfer dramatically increases print stability (see the [TPU Crocs](https://www.reddit.com/r/BambuLab/comments/1d0qeg1/lol_i_never_thought_id_see_the_day/)).
4. Keep this as an "information only" preview so the user can know how much theoretical savings they can get by adjusting their part. This could also be extended for different purposes, such as optimal orientation for mass manufacturing, optimal orientation for appearance, optimal orientation for minimal supports, etc. 

Most of these approaches require input from the user, and they're not perfect on their own. However, this is something new and would probably be iterated upon a few times with the different approaches anyway. Given that, the least risky option (in terms of print failures would be