OrcaSlicer/OrcaSlicer #7112: Bambu Printer camera no longer shows up/black with the occasional player is malfunctioning try reinstalling

firemanz8, 2024-10-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1 and 2.2.0-beta

### Operating System (OS)

Linux

### OS Version

Linux mint 21.3

### Additional system information

[image]

### Printer

Bambu Labs x1c and Ender 3 running klipper

### How to reproduce

when you open the orca slicer and navagate over to the device tab and click on the play button net to the camera it never loads
[image]
[image]

### Actual results

error message occurs with no view of camera

### Expected results

it should play thew camera live feed it was working properly with the stable build of bambu studio however i started to setup my ender 3 again so i want to use orca to use 1 slicer for both printers

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17372231/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

it works fine in the bambu studio application for linux
[image]

edward-p, 2024-10-27:
AppImage version works for me on Arch Linux, but self-build version doesn't.

tflucke, 2024-12-07:
I can verify that I have the same problem.

tuncerumut, 2024-12-30:
+1
Is this issue discussed anywhere else and solution suggested or just no solution yet?

HailGoosifer, 2025-01-02:
+1
This is still an issue for me as well.

lajborS, 2025-01-10:
Me too. I have the same issue on archlinux. Bambu studio works properly.

JRO119, 2025-01-19:
Same issue here.

OS: Linux
Printer: Bambu Lab P1S - firmware 01.07.00.00

Bambu Studio - camera working
    version 1.10.1.50
libbambu_networking.so
    version 01.10.01.01

Orca Slicer - camera not working:  "Player is malfunctioning. Please reinstall the system player. [2:2]"
    version 2.2.0 (build)
    2.3.0-dev (nightly flatpak)
libbambu_networking.so
    version 01.09.05.01

EchoJunction, 2025-01-25:
Same issue over here on Linux.

Printer: Bambu Lab P1S - firmware 01.07.00.00

OrcaSlicer 2.3.0-dev

ITankForCAD, 2025-01-27:
Seeing the same issue on Fedora 41 using kernel 6.12.9 with the 2.3 Dev Flatpak on a P1S with firmware 01.07.00.00. @Noisyfox Any potential fix or updates ? Since Bambu's rug pull, I've been looking to migrate my entire workflow to OrcaSlicer, but this is the only hiccup left.

stremlenye, 2025-01-27:
Same here on MacOS Sequoia 15.2.

jdecoste, 2025-02-01:
Same here. 
Fedora 41
Kernel 6.12.11
2.3 dev flatpak

X1 Carbon 01.08.00.00 w/ X1 Plus 2.0

LepreseanGit, 2025-03-16:
Same here, 
Mac OSx Sequoia
X1C

pinot911, 2025-03-20:
same for me, macosx sonoma

vpassapera, 2025-04-03:
Same here, manjaro, appimage version 2.3 stable. Printer is an X1C. Bambu studio works fine for camera (but everything else sucks compared to Orca).

Please fix this one issue!

closed 2025-09-28 by github-actions[bot]