OrcaSlicer/OrcaSlicer #7115: Spiral Vase seam on smooth objects / Spiral vase not following object contour

fritzw, 2024-10-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1 and 2.2.0 RC

### Operating System (OS)

Linux, Windows

### OS Version

Mint 22, Windows 11

### Additional system information

irrelevant

### Printer

irrelevant

### How to reproduce

- Add standard cone primitive to build plate
- Switch to vase mode
- Slice
- Notice seam-like artifact in preview and in print

### Actual results

Print shows a "seam-like" artifact on the smooth surface of the cone:

[image]

The problem becomes more severe for more severe overhangs (making the cone wider but not taller). Here's the same cone but scaled down from 25.6mm to 9mm height:

[image]

The root cause seems to be that the slicer cuts the object into horizontal slices as usual in the "normal" non-spiral mode, and then joins them afterward to create a spiral, instead of following the actual surface of the object in a spiral trajectory. This can be seen by overlaying a circle on the preview, which aligns perfectly with the contours of the toolpath. This should not be the case if the toolpath was an actual spiral ascending a cone:

[image]

Activating the "Smooth spiral" mode merely hides the fact by blending the seam over a larger distance. Nevertheless, most of the extrusion path is still circular and does not faithfully follow the contour of the object. Only about 1/3 of it actually approximates a spiral due to the smooth blending:

[image]

### Expected results

- There should be no seam in vase mode for a smooth object
- The toolpath should faithfully follow the contour of the object as Z increases, not follow the contour of the object at a single Z height.

### Project file & Debug log uploads

[orcaslicer_vase_mode_bug.zip](https://github.com/user-attachments/files/17375528/orcaslicer_vase_mode_bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

- Not sure if this this is the same as #4298 because that mainly deals with complex models. This issue occurs even with very simple shapes like a cone.
- PrusaSlicer appears to do the same "blending" of circular paths as the "Smooth Spiral" mode, as far as I can tell from drawing circles on Screenshots.

gmaiorov, 2024-12-21:
That still affects me on 2.2.0, and causes gap between bottom and start of the spiral, so bottom tries to fall off.

<details>
<summary>Details</summary>

[image]
[image]
[image]
[image]

</details>

fritzw, 2025-03-22:
Still relevant

menandore, 2025-07-07:
Still seeing this problem myself; it's very obvious if I choose a narrow line width

[image]

fritzw, 2025-10-06:
Not stale

fritzw, 2026-10-30:
> I've been having this same issue.
> I can either turn off smooth spiral and get a vertical seam, or turn it on and have the first part of the spiral stick out so far it becomes a weak spot that has ruined several prints in the past because there's not enough overlap with the layer below.

I'd guess that the start of the spiral sticking out is a different bug in the smooth spiral code that only affects the beginning.