OrcaSlicer/OrcaSlicer #7119: Injection molding while printing

RChadwick7, 2024-10-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The problem with printing layer by layer has always been that parts tend to break along layer lines. Also, it appears better looking prints can be achieved by printing the outside layers first.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Injection molded parts are always stronger, but what is injection molding? A hot end squirting hot plastic into a mold. This sounds a lot like a 3d printer. Assuming the printer has an adequate hot end, and likely a heated chamber, why not slice so that there are pockets created, then these pockets are filled with a stationary squirt of plastic from the hot end? Another variation would be to print thicker layers inside the print by going slower and extruding more.

vgdh, 2024-10-15:
[ Z-pinning to increase isotropy of part strength #4815 ](https://github.com/SoftFever/OrcaSlicer/discussions/4815)

Rickthebig, 2024-10-16:
I think what you are asking for is something similar to Cura's "slice as a mold" function, which is very interesting, as you can use any model to create a mold for it. You could, for example, pour wax into that mold and get your original model made out of wax.

mytechguyri, 2024-10-18:
I've seen this attempted, and the problem is, the filament cools too quickly to do this in the way you describe, but essentially what you're doing is creating solid infill.   This is already an option in the slicer.

closed 2025-01-25 by github-actions[bot]