OrcaSlicer/OrcaSlicer #7120: Add resonance avoidance feature from QIDI studio to orca slicer

Qjeezy, 2024-10-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There are certain machines that can benefit from having a resonance avoidance speed feature to prevent unwanted stepper motor noise due to certain speed settings. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It should mimic the resonance avoidance feature from QIDI studio exactly. You input a range of speeds that your printer does not play nice with and it will do its best to avoid those printing speeds. 

### Describe alternatives you've considered

Alternatives would be to manually set all speeds accordingly which may not even be effective at avoiding certain speeds

xokahag, 2024-10-19:
Under filament settings > cooling you can find the option "don't slow down outer walls", so you can manually set the speed your printer works best at. Be careful though, it currently has higher priority then enforcing minimum layer times. See https://github.com/SoftFever/OrcaSlicer/issues/6859

The resonance causing VFA on the walls is mostly [Motor resonance ripple](https://www.youtube.com/watch?v=-et5eMyLlUs) in my experience, so it depends on each motors individual speed , and *not* the total movement speed. On a core XY printer, both motors have the same absolute speed `√|v|` when moving in X or Y direction. When moving diagonally, only one motor turns with the absolute speed `|v|`. Logically for all angles in between the extremes, the motor speed varies between 0 the maximum.

To avoid resonance it would therefore be a better idea to avoid certain motor speeds instead of movement speeds to minimize VFA artifacts.

PS:
I looked up some QIDI prints with VFA issues and it seems more pronounced then on my non QIDI printer. I'd be curious how your prints look like. Can you do a VFA test print? You find it under Calibration > More... > VFA; for settings I usually choose 50mm/s to 500mm/s with 50mm/s step, adjust max speed to whatever your printer, filament and profile can handle.

Qjeezy, 2024-10-19:
> Under filament settings > cooling you can find the option "don't slow down outer walls", so you can manually set the speed your printer works best at. Be careful though, it currently has higher [...]

I actually do have the don't slow down outer walls enabled already and it does not help with avoiding certain speeds, even with the speed settings adjusted. 

I have ran the Klippian shake & tune on my QIDI plus 4 and the main no go zone is between 62 mm/s and 110mm/s. Even with don't slow down outer walls and my speeds set to the proper speeds to match the green zones, the filaments max volumetric speed setting overrides everything and certain features of the object prints at unwanted resonance causing speeds. 

My VFA's aren't actually bad at all. Nearly non existent, but the stepper motors scream within the no go speed zones shown on the shake & tune results. 

Slicing the same exact file with the same settings in QIDI studio, with the resonance avoidance feature speeds set to avoid 60-110mm/s successfully changes the print speeds in the sliced preview so that the motor resonance noises will not occur. 

This feature for me is more so about the noise and easily avoiding the speeds that create the noise and not so much about the VFA, but it controlling and minimizing VFA is a great side effect. 

In conclusion, the resonance avoidance feature from QIDI studio is a super simple and easy way to set a speed range that you do not want your printer to use during printing and it seems to override any other speed settings to achieve these results.

xokahag, 2024-10-19:
>This feature for me is more so about the noise and easily avoiding the speeds that create the noise and not so much about the VFA, but it controlling and minimizing VFA is a great side effect.

I get it now, wasn't really clear from the initial post.

>Even with don't slow down outer walls and my speeds set to the proper speeds to match the green zones, the filaments max volumetric speed setting overrides everything and certain features of the object prints at unwanted resonance causing speeds.

To to make sure: The max volumetric flow rate prevents your printer to print at speeds higher then then 110mm/s, so in these cases the slicer should slow down to 60mm/s, if possible?

Qjeezy, 2024-10-19:
> I get it now, wasn't really clear from the initial post.
>
>
>
>
>
>
> To to make sure: The max volumetric flow rate prevents your printer to print at speeds higher then then 110mm/s, so in these cases the [...]

The max volumetric speed setting will throttle printing speeds equal to or less than your preset speed settings. For example, if your outer wall speed is set to 110 and your max volumetric speed is set to 12, if printing at 110mm/s requires more flow than 12, the printer will slow down that outer wall speed to an appropriate speed less than 110 in order to maintain that 12 or less volumetric flow. 

So in this case, the printing speeds for that outer wall is whatever the slicer thinks it should be. 

Without the suggested resonance avoidance feature, it can place that speed into the range of speeds you want to avoid. 

With the resonance avoidance feature set to a range of 60-105, if the slicers max volumetric speed setting has to slow down that 110 outer wall speed and it winds up in that range, it will drop it down straight to 59 to avoid that range. 

Hope that makes sense lol.

Mikepyt87, 2024-11-06:
I too would like to see this feature brought over to Orca! The resonance avoidance tool in QIDI Studio is one of those subtle but incredibly useful options, and bringing it over here would be a real win for a lot of us. It’s about more than just keeping the noise down (though that alone is huge!); it’s also about getting around those frustrating moments where the slicer, despite all our tweaks, still ends up running in that “no-go” speed range.

Without this feature, manually setting speeds can be like playing whack-a-mole with resonance—try one setting, it’s too noisy, try another, still not quite there. Having a way to specify a range we know is problematic and just skip it altogether would simplify things a ton. It’s not just about avoiding noise but also keeping print quality consistent.

I think this is one of those upgrades that would add a lot of value for many users, and I’m excited about the possibility. Thanks for considering it!

yayuuu, 2025-01-08:
I'm also looking for this feature. I'd like to print inner walls at speeds higher than 340mm/s if possible and outer wall at less than 260mm/s. I have a horrible resonances at around 300mm/s. If I set these speeds however, volumetric flow can throttle my speed to around 300mm/s with certain layer heights. In this case I'd like the slicer to throttle it even more, to 260mm/s, by excluding speeds between 260-340mm/s as possible speeds. Either go above 340mm/s if possible or below 260mm/s if not possible. 
It would be nice to be able to manually specify excluded speed ranges in the printer settings. (more than one range if possible)

Qjeezy, 2025-01-08:
> I'm also looking for this feature. I'd like to print inner walls at speeds higher than 340mm/s if possible and outer wall at less than 260mm/s. I have a horrible resonances at around 300mm/s. If I [...]

This is precisely what I want as well. Having the ability to set more than 1 range of speeds is an awesome suggestion.

edspeds, 2025-02-01:
Trying to keep this alive as it's something that would be great for my use as well.  My toolchanger also gives bad VFR between 60 and 110 mm/s so I'd like to be able to tell the slicer to make those speeds off limits.

lindnjoe, 2025-02-13:
Add me to the list as well :)

Squallzz, 2025-02-18:
Bump bump bump we need this

d33pm4n, 2025-02-18:
How about handling such "no-go" ranges like the height modifiers?
Imagine having possibility to set not one but multiple speed ranges to avoid.
Example exclusion list:

- from 60 mm/s to 75 mm/s;
- from 105 mm/s to 120 mm/s;
- from 260 mm/s to 300 mm/s.

Could be a bit confusing for newer users, but very usefull once mastered.

Qjeezy, 2025-02-18:
> How about handling such "no-go" ranges like the height modifiers?
> Imagine having possibility to set not one but multiple speed ranges to avoid.
> Example exclusion list:
>
> - from 60 mm/s to 75 mm/s;
> - [...]

Having the ability to effectively disable multiple speed ranges would be useful for sure. This way we could totally avoid all speed ranges that align with resonance peaks.

edspeds, 2025-02-19:
> How about handling such "no-go" ranges like the height modifiers? Imagine having possibility to set not one but multiple speed ranges to avoid. Example exclusion list:
>
>     * from 60 mm/s to 75 [...]

Hopefully they see the value in this as I'm of the opinion the majority of print artifacts can be attributed to speed resonance.

Qjeezy, 2025-02-23:
@SoftFever this would be a very nice feature to have and it would help just about every klipper printer out there. Should be an easy port from QIDI studio.

d33pm4n, 2025-02-26:
> How about handling such "no-go" ranges like the height modifiers? Imagine having possibility to set not one but multiple speed ranges to avoid. Example exclusion list:
>
>     * from 60 mm/s to 75 [...]

Even better - it could be controlled via same type control field like in "Flow Compensation Model" which pops up when checking "Small area flow compensation" in Quality tab.
Might be more simplistic?

All in all, it is down to array of ranges to avoid. Hopefully it will be a simple port as Qjeezy said and then extending it to multi range.

linouxis9, 2025-04-04:
Bumping too!