OrcaSlicer/OrcaSlicer #7124: Zero bed temperature on dual filament print

DimitryNechaev, 2024-10-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.2.0 RC Release

### Operating System (OS)

Windows

### OS Version

Windows

### Printer

Snapmaker J1

### How to reproduce

I have sliced existing project that worked. Just wanted to duplicate items to print more at once.
It is a PLA + TPU print, in the previous version it would set bed temp to 60 degrees.
In the printer file selection preview screen, it reports bed temp 60 for the print but when prints it starts with the cold bed

I have switched on the original printer profile, no difference

### Actual results

I have changed the printer started GCode to dump values.

```
; --- initial_extruder: [initial_extruder]
; --- has_wipe_tower: [has_wipe_tower]
; --- total_toolchanges: [total_toolchanges]
; --- T0: {is_extruder_used[0]}
; --- T1: {is_extruder_used[1]}
; --- bed temp initial extruder {first_layer_bed_temperature[initial_extruder]}
; --- bed temp extruder 0 {first_layer_bed_temperature[0]}
; --- bed temp extruder 1 {first_layer_bed_temperature[1]}
```

The result comes out as following:
```
; --- initial_extruder: 0         
; --- has_wipe_tower: 1          
 ; --- total_toolchanges: 91      
 ; --- T0: true                   
 ; --- T1: true                    
; --- initial extruder 0          
; --- bed temp initial extruder 0 
; --- bed temp extruder 0 0      
 ; --- bed temp extruder 1 0
```

### Expected results

Bed temp for extruder 0 should be 60 and for extruder 1 should be 40

### Project file & Debug log uploads

Project file:
[sample project.zip](https://github.com/user-attachments/files/17386768/sample.project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

No issue observed printing a single filament

hliebscher, 2024-10-16:
for me it woks fine

; --- initial_extruder: 1
; --- has_wipe_tower: 1
; --- total_toolchanges: 5
; --- T0: true
; --- T1: true
; --- bed temp initial extruder 59
; --- bed temp extruder 0 60
; --- bed temp extruder 1 59

hliebscher, 2024-10-18:
@DimitryNechaev @SoftFever 
pls close it works

DimitryNechaev, 2024-10-24:
@hliebscher Please help me understand what to change, as I believe it worked in the previous version. I suspect it may something to do with the updates for J1S included into the release.

When clicking on the bed properties it shows the bed as expected, although bed type is locked:
[image]

There is no property to edit bed type:
[image]

And both filaments are non-0, although their bed types mismatch with the bed type:
[image]
[image]

So there are 2 questions:
How on earth do I set filament bed temp for "textured bed"?
And how do I change the bed type?

DimitryNechaev, 2024-10-24:
Also I found how to allow bed type selection on the printer
[image]

Unfortunately the selection makes no effect to the bed - it still stays textured and the selection is locked

It is a stopper. How an ordinal user is able to figure this out?

hliebscher, 2024-10-25:
at first set 
[image]

then go to filament and set bed temp at your selected build plate
[image]

closed 2025-01-28 by SoftFever