OrcaSlicer/OrcaSlicer #7127: Accel to deccel - min_cruise_ratio not working as spected

JuanR3D, 2024-10-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-BETA

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Custom

### How to reproduce

1. Activate acel to decel parameter in velocity settings of process
2. Set accel_to_deccel 1%
3. Set diferent acelerations in travel and other movements

### Actual results

This places in the gcode the following commands:
SET_VELOCITY_LIMIT ACCEL=xxx ACCEL_TO_DECEL= (xxx*0.01)
for example, when commanded accel is 3000, accel_to_decel=30

I dunno if this has been implemented like this on desire when klipper migrated this parameter to min cruise ratio, but what this is actually doing is setting all the time my min cruise ratio to the max: 0.99 or giving me an error.

For example, when this is commanded: SET_VELOCITY_LIMIT ACCEL_TO_DECEL=30 my min_cruise_ratio goes to 0.99 but when this is commanded: SET_VELOCITY_LIMIT ACCEL_TO_DECEL=10 , it goest to 1 and klipper gives an error because this value should be between 0 and 0.99

### Expected results

Maybe i'm incorrect, but i think accel_to_deccel or min_cruise_ratio should not depend of the accel commanded but be a static value.

As i understand, this parameter configures if the velocity profile comes to a triangular or trapezoidal speed, depending of how much min_cruise_ratio you desire.

In my case, i want to create various profiles, and i want to make one of them faster, so i want to set this parameter to 1% so the machine can make triangular velocity profiles if required (all time accelerating, no cruise velocity required in the movement, just reach it)

In this way, i think that if i set this accel_to_deccel to 1%, i just should send  the following command when starting the process or object with its respective value.

Also, it should use SET_VELOCITY_LIMIT MINIMUM_CRUISE_RATIO=value instead of SET_VELOCITY_LIMIT ACCEL_TO_DECEL=value as it doesn't work anymore in klipper

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2024-11-03:
Set it to 100% to achieve what you want - no crusing.

igiannakas, 2024-11-03:
By the way, this command still works on klipper - it has been deprecated in macros and the UI but gcode is processed correctly: SET_VELOCITY_LIMIT ACCEL_TO_DECEL=value.

The issue comes from converting that 1% value. Its too low. For what you want to achieve, which is no cruising you should set the value in Orca to 100% or disable it completely in Orca and set the min cruise ratio in klipper to 0.

JuanR3D, 2024-11-04:
Hi @igiannakas , thanks for response!
And yes, what i want to do is to create a fast printing mode where there is no cruising... 

As i have different print modes (i only modify this parameter in this fast mode)... How can i insert this command in orca in order to disable cruise only in this fast mode?

igiannakas, 2024-11-04:
Just set accel to decel in the orca profile to 100%