OrcaSlicer/OrcaSlicer #7135: Failed to get AMS mapping table 07ff 8012 002617

WimVDK, 2024-10-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-RC

### Operating System (OS)

Windows

### OS Version

Win 11 23H2

### Printer

X1C

### How to reproduce

1. Open project
2. Use custom filaments ABS-GF (slots 15 & 18) (is just for slicer use purposes-
3. Send to printer, Slot 18 gets properly detected as AMS slot D4

### Actual results

Printer heats up and after minutes just sits there and eventually shows above error code

### Expected results

Not show that error and proceed to printing

### Project file & Debug log uploads

[babo_duo_cover-plate.zip](https://github.com/user-attachments/files/17399721/babo_duo_cover-plate.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

closed 2025-01-23 by github-actions[bot]

su-sd, 2025-05-11:
I also get this error with a P1S on the latest firmware.
Very likely related: https://github.com/SoftFever/OrcaSlicer/issues/8063

WimVDK, 2025-05-11:
That's a different issue. The latest P1S firmware this is a known bug. Roll back