OrcaSlicer/OrcaSlicer #7136: Textured PEI Bed Temperature wrong in every included Prusa XL Filament Profile

corefile, 2024-10-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0-rc

### Operating System (OS)

macOS

### OS Version

MacOS 15.0.1 

### Printer

Prusa XL 5 Toolhead

### How to reproduce

1. Enable "Support Multi Bed types" in the printer properties
2. Select "Textured PEI Plate" for Build Plate drop down
3. Edit any of the include Filament Profiles
4. Scroll down to bed temperature and observe incorrectly set to 45 degress for Textured PEI Plate

### Actual results

Prints will fail as its to low for bed adhesion

### Expected results

If the temperature is set correctly for the filament type you will have proper bed adhesion

### Project file & Debug log uploads

[BadProfile.3mf.zip](https://github.com/user-attachments/files/17402618/BadProfile.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

avxmw, 2024-11-05:
Same problem here with Anycubic filament profiles.

cochcoder, 2025-03-12:
I don't see a setting in Pursa Slicer for what the temp should be on PEI bed sheets. Where could I find the correct temperature to set it to in the profile?

closed 2025-06-19 by github-actions[bot]