OrcaSlicer/OrcaSlicer #7149: [Gentoo] Nightly AppImage: /tmp/.mount_orca_sIlU9GN/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

julie-de-ville, 2024-10-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

Current nightly

### Operating System (OS)

Linux

### OS Version

Gentoo, Linux v6.6.52

### Additional system information

Thinkpad t450s

### Printer

Ender 3 S1

### How to reproduce

Download and execute nightly appimage, I have installed webkit globally.

### Actual results

Orca does not start, and returns `/tmp/.mount_orca_sIlU9GN/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory`

### Expected results

Orca Slicer runs without error

### Project file & Debug log uploads

/tmp/.mount_orca_sIlU9GN/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I would really like to not need to install webkit in its entirety.

el-quique, 2024-10-26:
Hi try this:
sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18

And create "start_orca.sh"
#!/bin/bash
WEBKIT_DISABLE_DMABUF_RENDERER=1 /opt/OrcaSlicer/OrcaSlicer.AppImage

dogs-1985, 2024-11-03:
> Hi try this: sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37 sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 [...]

Hi,

Is this intended to be a temporary fix, or is there any plan for the developers to address this issue in the future releases of OrcaSlicer?

Norbiros, 2025-01-22:
There are multiple WebKIT releases available on gentoo: `6`, `4.1` and `4`. So to install version needed for orca slicer use `emerge --ask net-libs/webkit-gtk:4` and it should work. The only possible fix that developers can make is just to update to newer webkit versions (for example Tauri framework did that).

closed 2025-04-30 by github-actions[bot]

KingJames0000, 2025-06-01:
I'm still seeing this issue. I can't get that library on Gentoo as far as I can tell. I can get webkit-gtk, but Gentoo is still only using a version 2 in stable, I believe, and I can't get version 4 without a lot of hassle I don't care to do to get this slicer.

This is the Achilles heal of AppImages, in my opinion. Developers assume if it runs on the most used distros (like Debian & Ubuntu & Fedora), it will work on every distro, so they don't check to make sure they include all libraries. I wish Orca's team would add this library to the build, so we Gentoo users can partake! Or, at least, make a Flatpak as an alternative.