OrcaSlicer/OrcaSlicer #7150: Filament Edits are NOT saved when modified

grizzlybus, 2024-10-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

15.0.1

### Additional system information

Macbook Pro - 2023 16"

### Printer

Voron 2.4

### How to reproduce

Open OrcaSlicer
Open New Project
I have a user filament that is set to filament type "PLA"
I try to update the filament to "ABS"
Click Save (disk icon by the Filament) 
Quit
Open OrcaSlicer 
Filament is still PLA when I reopen a new project and try to edit my filament settings

### Actual results

Does not save changes to filament edits

### Expected results

This is the filament I want to edit:
[image]
This is the type (PLA):
[image]
I click the Disk icon at the top, the star disappears after I click OK:
[image]
[image]
Double check the save:
[image]
Quit, restart and...
[image]
No changes saved

### Project file & Debug log uploads

[logs.zip](https://github.com/user-attachments/files/17415089/logs.zip)
[badproject.zip](https://github.com/user-attachments/files/17415110/badproject.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

tlhintoq, 2024-10-20:
When you save, you have to save with a new name.  The new material inherits from the base material.
Trying to trick it by using the same name doesn't play.
Just give it a new name by switching things around.  Instead of 
Polymaker ABS Red
try
ABS Polymaker Red

Then all your materials will list together, and then together by brand
ABS Polymaker Blue
ABS Polymaker Gray
ABS Polymaker Green
ABS Polymaker Red
ABS Sunlu blue
{...}

closed 2025-01-27 by github-actions[bot]

avispa4291, 2025-03-06:
I am having the same exact problem since updating to 2.2, it will not save any new filaments and I also can't add custom fliaments