OrcaSlicer/OrcaSlicer #7152: Add support for "Non-Planar Top Layer Fuzzyskin"

misterjmw, 2024-10-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It would be nice to have a toggle-able feature to support fuzzy skin on non-planar, top surfaces.  

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

A dev has created and tested this feature with several slicers and is open source. It would be nice to see an integration of the functionality from https://github.com/TengerTechnologies/Fuzzyficator as an Orcaslicer feature.  Discussion on the project can be found [here.](https://old.reddit.com/r/3Dprinting/comments/1g5pd5e/update_nonplanar_top_layer_fuzzyskin_now/). The feature should work with Marlin and Klipper. 

### Describe alternatives you've considered

The only alternative I'm aware of is GCode post-processing, which is a nuisance.

kdegeek, 2024-10-17:
You beat me to this by 13 mins haha

Rickthebig, 2024-10-17:
"The feature should work with Marlin and Klipper"
What about RRF? Something against implementing it also to work with RRF?
Or just completely agnostic of the gcode flavor used?

misterjmw, 2024-10-17:
> "The feature should work with Marlin and Klipper" What about RRF? Something against implementing it also to work with RRF? Or just completely agnostic of the gcode flavor used?

The form only asks about Marlin and Klipper. Anything outside of that, I have no/too little experience to provide an answer.  I would say generally that exclusion should not be taken as a statement that "It will not work with ________".

vgdh, 2024-10-18:
Have you tested it? How is the quality? I would love to see some photos.

misterjmw, 2024-10-18:
> Have you tested it? How is the quality? I would love to see some photos.

There are photos and videos in the discussion topic, and a youtube video here:
https://www.youtube.com/watch?v=EhKnqHey6l8

DVSVIDEO, 2024-10-20:
Very good idea, it would be nice to have such a feature.
Something similar already existed: a sawtooth pattern in the Superslicer. It didn't work like that, but instead of a smooth surface it made this ridge

PhilBaz, 2024-10-23:
[youtube.com/watch?v=EhKnqHey6l8](https://www.youtube.com/watch?v=EhKnqHey6l8)

Fuzzy top skin. Obviously this is amazing.

PhilBaz, 2024-10-23:
@DVSVIDEO Ohh, that's super cool too!

TengerTechnologies, 2024-10-25:
I will make some more testprints. I don't think it's ready for an implementation yet.

TengerTechnologies, 2024-10-25:
> "The feature should work with Marlin and Klipper" What about RRF? Something against implementing it also to work with RRF? Or just completely agnostic of the gcode flavor used?

It currently is only tested with Marlinflavour

vgdh, 2024-10-25:
> > "The feature should work with Marlin and Klipper" What about RRF? Something against implementing it also to work with RRF? Or just completely agnostic of the gcode flavor used?
> 
> It currently is only tested with Marlinflavour

It doesn't work with orca gcode, because orca issues z coordinate in every command on top layer https://github.com/SoftFever/OrcaSlicer/issues/7207 and also script uses prusa stile of layer type tips (don't remember exactly but it something like ;Type top layer) so little modification to the script can fix it.

TengerTechnologies, 2024-10-25:
> It doesn't work with orca gcode, because orca issues z coordinate in every command on top layer https://github.com/SoftFever/OrcaSlicer/issues/7207 and also script uses prusa stile of layer type [...]

To implement it in orca slicer I would do it in the gcode generation, not as postprocessing. So the script is not usable anyways. 

A postprocessing script for orca is nearly ready for release. I just have to further test it.(Probably done and commited by the end of the weekend)

igiannakas, 2024-11-09:
> To implement it in orca slicer I would do it in the gcode generation, not as postprocessing. So the script is not usable anyways.
>
> A postprocessing script for orca is nearly ready for release. I [...]

I’ve been intrigued by this and am keen to start implementing this in orca.

im wondering though why you mention doing this in the gcode generation step and not as a post processor? 

I think the gcode generation step assumes planar printing vs the post processing step when slicing, where we can do whatever we like - plus the script is pretty straightforward to implement as an integrated post processor (like the fan speed mover, pressure equalizer and adaptive PA which all are post processors).

TengerTechnologies, 2024-11-09:
> I’ve been intrigued by this and am keen to start implementing this in orca.
>
>
>
> im wondering though why you mention doing this in the gcode generation step and not as a post processor?
>
>
>
> I think [...]

I was thinking that it may be easier to work with support blockers and paint on. But I may be weong though.

However I'm already working on a pull request and I would kind of love to implement it myself as I put so much work into the idea. Or at least work together on it. If you understand this and if you are interested in working together you could write me on reddit u/TenTech_YT

igiannakas, 2024-11-09:
Awesome! Wasn’t sure if you were working on it, but since you are please do continue :)

Happy to review the code and help out with testing when it’s done :)

igiannakas, 2024-11-09:
Safety wise it should be fine as the moves you’re doing are a layer tall. My suggestion would be to not decrement Z but rather mostly aim to increment it. This way you’ll be safe from any potential collisions.

TengerTechnologies, 2024-11-09:
> Safety wise it should be fine as the moves you’re doing are a layer tall. My suggestion would be to not decrement Z but rather mostly aim to increment it. This way you’ll be safe from any potential collisions.

Yeah I think it shouldn't be much of a problem because z-hop isn't a problem either. I just haven't seen someone running the script on a corexy and I don't know how much wear it will create on the Z axis if you move the whole builtplate. 

I was thinking to remove the Z-min parameter completely and always leave it on 0, but then again I thought why not give the ability to play around with it.

Another problem will be the ironing. It kind of has to be done non-planar too, because otherwise there will be collisions.

TengerTechnologies, 2024-11-09:
> Awesome! Wasn’t sure if you were working on it, but since you are please do continue :)
> 
> Happy to review the code and help out with testing when it’s done :)

Thank you! I will surely reach out to you when I'm stuck or need someone to review the code :)

igiannakas, 2024-11-09:
I wouldn’t enable it when ironing is set- it kind of defeats the point of fuzzy skin on top layers. You can simply disable the post processor in the code and set the parameter to toggle off when ironing is set in the config manipulation code.

take a look at my pressure equalizer Pr for example on how to toggle configs: https://github.com/SoftFever/OrcaSlicer/pull/2161

misterjmw, 2025-02-08:
Thanks Stale-Bot, but this request is only stale because Orca/Devs haven't made movement.

FasteryXD, 2025-06-02:
Is there progress?

Chavife, 2026-09-02:
Hi everyone,

  I've built a proof-of-concept top surface fuzzy skin for OrcaSlicer, integrated natively into the slicer rather
  than bolted on afterwards. Fair warning up front: this was written with heavy AI assistance, so treat it as a
  starting point rather than production code.

  It was inspired by Fuzzyficator (https://github.com/TengerTechnologies/Fuzzyficator), which does this as a
  G-code post-processing script. The idea seemed worth having inside the slicer, so that's what this is.

  What it does

  The wall fuzzy skin generator can't reach top surfaces — it works by displacing points sideways along the
  surface normal, and a horizontal surface has no sideways. So this adds the two mechanisms that do work on a flat
  surface, matching the wall modes:

  - Displacement — moves the nozzle up and down while printing the top solid infill. The most visible relief, but
    it adds Z motion to every segment and slows printing down noticeably.
  - Extrusion — Z stays flat and the deposited volume is modulated instead. No extra motion, no speed penalty,
    subtler matte finish rather than raised relief.
  - Combined — half the amplitude from each.

  A detail I think is worth highlighting: it samples the same noise field as the walls, at (x, y, slice_z). That
  means the texture is continuous where a top surface meets a wall, rather than looking like two separate effects
  that happen to meet at an edge. A post-processing script fundamentally can't do this, because by then the noise
  field is gone.

  It also composes with Z Anti-Aliasing — displacement is added to the contoured Z, so a ZAA'd top surface keeps
  following the mesh and gains texture.

  Settings

  Fully exposed in the UI under Print Settings → Others → Fuzzy skin:

  - Fuzzy skin on top surfaces — Disabled / Displacement / Extrusion / Combined
  - Top surface parameters — Same as walls / Custom
  - Top point distance and thickness (when Custom)

  All noise generators work except Ripple (see below).

  Known limitations

  - Paint-on fuzzy skin is not supported. Painted regions don't get top surface fuzz — the setting applies
    globally. I've verified this doesn't work; it's probably the first thing worth fixing.
  - Ripple noise isn't supported on top surfaces. It's generated along a wall loop's arc length and displaces
    along the path normal, neither of which means anything on a flat surface, so it falls back to Classic.
  - Noise type, scale, octaves and persistence are always shared with the walls. This is deliberate — it's what
    makes the texture line up — but it does mean you can't give the top its own character.
  - Ironing will flatten the fuzz if you have both enabled. Needs a warning or an interlock.
  - The 50/50 amplitude split in Combined mode is an untuned guess.
  - In Displacement mode, extrusion is scaled by how far the nozzle dips, so the downward half is limited to keep
    flow high enough for a continuous bead. Above roughly three quarters of the layer height, the texture grows
    upward only.

  I built this for myself, but it seems like something the community would benefit from. It's far from perfect and
  needs polish, but it works well in my testing. Please take the code, contribute, or change whatever's needed to
  get it release-worthy.

  There is quite a bit of certainty that I will be doing some improvements. Also I am up for contributing although I don't have that much of my time unfortunately.

  Everything is here: https://github.com/Chavife/OrcaSlicer/tree/fuzzy-skin-top-surfaces

[image]

[image]

[image]

[image]

[image]

Felix14-v2, 2026-09-02:
> Everything is here: https://github.com/Chavife/OrcaSlicer/tree/fuzzy-skin-top-surfaces

Do you plan sending a PR once the code is ready?

Chavife, 2026-09-02:
I can open a WIP PR if that helps. The thing is that I would love to collaborate on this with someone who understand Orca's libslic3r better. The last thing I want, is to introduce a nasty regression.
As I wrote most of the code is Vibe coded, since I don't have that much time, to tinker around. Although as I read through the code it is in quite good shape to just polish it up.

Chavife, 2026-09-02:
Well I guess here we go: 
https://github.com/OrcaSlicer/OrcaSlicer/pull/15500