OrcaSlicer/OrcaSlicer #7154: Error Pushing Prints using Upload Button in OrcaSlicer

JaredMichael-ENG, 2024-10-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.2.0-beta2

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Qidi Plus 4

### How to reproduce

1. Slice Plate
2. Click Print
3. Click Upload and Print

### Actual results

[image]
I can bypass this by exporting the G code file and then traveling to the device tab to upload the raw gcode to the jobs tab and starting the print.

### Expected results

Normally, This will send the G code directly to the printer and the printing will commence.

### Project file & Debug log uploads

[snapon_Logo.zip](https://github.com/user-attachments/files/17420588/snapon_Logo.zip)

[debug_Thu_Oct_17_12_52_32_26056.log.0.zip](https://github.com/user-attachments/files/17420633/debug_Thu_Oct_17_12_52_32_26056.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

william-orange, 2024-12-25:
Same problem here but only on one machine!

DoomedSouls, 2025-01-15:
I have the same problem but the problem only appeared after I created an account and password on the Fluidd web interface. Fluidd in OrcaSlicer works after I have logged in. But I can't send a print job directly because the error message mentioned in the first post comes up.

bill-orange, 2025-01-15:
@DoomedSouls try entering the machine’s IP address in the OrcaSlicer network dialog box instead of fluids.local(or whatever yours is)

DoomedSouls, 2025-01-15:
I am an idiot. I just had to add the API key that is displayed in the Fluidd web interface under “User management” to the Physical printer window. Now I can upload the G-code without any problems.
[image]
[image]

william-orange, 2025-01-15:
Don’t feel bad. I had the same problem. Sent from my iPhone

IMG_20250115_201115.png (view on web)
IMG_20250115_201331.png (view on web)