OrcaSlicer/OrcaSlicer #7155: Modifier cuts apart the model

DVSVIDEO, 2024-10-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer V2.2.0 RC Release and earlier

### Operating System (OS)

Windows

### OS Version

Win10

### Additional system information

Not relevant

### Printer

Micromake C1 modified core xz printer

### How to reproduce

1. Load an model
2. add modifier e.g. cylinder
3. change the modifier body wall speed, or the number of walls, or the line width, or the wall loop direction etc.
4. slice the model
5. you get the segmented modell

### Actual results

Adding a modifier and some of its settings will cut the model incorrectly. These settings are not exhaustive:
- Changing the speed
- Changing the number of walls
- Changing the number of wall loops
- Changing the number of loops to change the number of loops
Recently, in v2.2.0-beta2, a bug was fixed where the modifier would cut the model if the fuzzy skin was turned on.
This bug is actually the same bug. These are still to be fixed. https://github.com/SoftFever/OrcaSlicer/issues/2647

[image]
[image]
[image]

### Expected results

If I want a model to print quickly, but I want good quality on the front visible part, I need to add a modifier where I can set it to print slower on some parts (or if I want a stronger part, to print more walls on that part).
This modifier should not generate unwanted walls inside the body, should not cut the model...

### Project file & Debug log uploads

[modifier issue.zip](https://github.com/user-attachments/files/17421570/modifier.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

xokahag, 2024-10-19:
All modifiers cut the models into regions, which get processed individually. For options this makes sense, eg. if you want a different material or different infill pattern, there is no way for the slicer to connect extrusion in one region to the extrusion in the other region. This also applies to wall count, wall direction, and wall print order.

However I *do* agree the speed settings in particular should be applied after the paths have been calculated, so that you can speed up or slow down specific parts.

To get more walls on a specific area, consider using a small region with infill set to "concentric" and "100% infill".

DVSVIDEO, 2024-10-20:
> All modifiers cut the models into regions, which get processed individually. For options this makes sense, eg. if you want a different material or different infill pattern, there is no way for the [...]

This is not true. Splitting into regions is fine, but the error is that it generates a wall for these regions within the model and splits it in two.
Until recently, this bug was also present in the Fuzzy skin setting, which has been fixed in the new version.
But if it could be fixed, it should be fixed for the other settings (speed, wall order etc....)This is not true. Splitting into regions is fine, but the error is that it generates a wall for these regions within the model and splits it in two.
Until recently, this bug was also present in the Fuzzy skin setting, which has been fixed in the new version.
But if it could be fixed, it should be fixed for the other settings (speed, wall order etc....)

Noisyfox, 2024-10-21:
Not every options are compatiable with each, especially options that changes the wall numbers. The program is designed on top of the assumption that EVERY walls are closed loops (except aarchne gap fills when wall is too thin).

Though I do agree that any option that do not affect the wall count should not make extra walls between regions.

DonutCables, 2024-10-22:
I can confirm this behavior is present even when the only change is Outer wall speed or Inner wall speed.

Splarkszter, 2024-11-11:
> Though I do agree that any option that do not affect the wall count, or does not break wall loops (such as wall speed) should not make extra walls between regions.

Understood. Yeah, i solved my issue of extra walls by using 100% concentric infill.
It would be good to have documented what settings may cause the model to be split into regions so we can avoid them if we need to.

Noisyfox, 2024-11-11:
> It would be good to have documented what settings may cause the model to be split into regions so we can avoid them if we need to.

Yeah that can definitely be improved.
At the moment, thise could be found here https://github.com/SoftFever/OrcaSlicer/blob/1088d0a6c713250ee3d14da791e27b76b4339363/src/libslic3r/Layer.cpp#L169-L195

eDaterKenobi, 2024-11-30:
did we ever find a fix for this? currently setting up modifier cubes to change wall thicknesses, bridge directions, and speeds, seems the only modifier that DOESNT isolate the modifier from the model is infill change.

RagingRoosevelt, 2024-12-23:
> The program is designed on top of the assumption that EVERY walls are closed loops and is printed as a continues extrusion (except aarchne gap fills when wall is too thin).

I think this assumption should be able to hold even when increasing or decreasing the wall loop count, without fully separating the wall loops.  For example
[image]
The alternative route shown here would preserve the assumption that all loops are closed, without adding a visible seam on the outside of the model, and without adding those four extra walls in the middle of the model.

eDaterKenobi, 2024-12-23:
> I think this assumption should be able to hold even when increasing or decreasing the wall loop count, without fully separating the wall loops. For example [...]

i dont fully understand the example. the "existing" diagram fully explains whats happening. but running any modifier cube with any wall count other than whats set of the main model will always output what is shown as "existing". im not understanding how to replicate the "+1 wall loop" nor "alternative" diagram. 

but i did fix it by running a height range modifier and setting the modifier to 1 wall, 0 tops, 100% infill. this creates all solid layer without any wall interruptions. or something like that. i did it 3 weeks ago but i do use the method i just explained to get smooth top layers when there are objects on top of top layers. hard to explain ill just upload an image 
[image]
[image]
[image]

first image is before. and then the 2nd image is with the modifer, showing one smooth top layer without interrupting. and then third image is just the next layer up. indicating the walls on top of the previous layer. rather than creating gaps between the walls and the layers below.

RagingRoosevelt, 2024-12-23:
The "+1 wall loop" shows how the default unmodified model is set up, and the "alternative" isn't currently possible but is my proposal for how to maintain the "closed loop" assumption while not adding more interior walls than is minimally necessary.

I'm not understanding your fix.  [Here's an example file](https://www.dropbox.com/scl/fi/ddu1n5fvw3ke55vmmra1f/wall_loop_example.3mf?rlkey=ombylzhtufq3005lpg6wiujsp&dl=0) showing the issue at hand.

[image]
[image]

How would one go about changing that so that it doesn't have all those inner loops?  My proposal would be like this, from a slicer algo perspective:

[image]

eDaterKenobi, 2024-12-23:
ur right. my fix was for another problem with orca slicer where it creates gaps below top layers. rather than making them solid. 
the only fix i have right now is to run 100% infill. changing the speed or wall count will always disrupt the original model's structure. 
[image]
[image]
[image]
[image]

top 2 images are current result
bottom 2 images are a workaround. creating a shape of the object u want to effect and setting it to 100% infill is the only option i can think of. even creating that portion in CAD would probably be better for curved or more complex geometry and then importing it as a modifier object.

RagingRoosevelt, 2024-12-23:
Yeah...

For me, this issue came up where I wanted a region with fewer wall loops, rather than more wall loops.  So for my application, the 100% infill trick isn't viable.

DVSVIDEO, 2025-01-02:
And wouldn't that be a good solution?
Neither the outer nor the inner wall would be cut by the modifier:

[image]

Splarkszter, 2025-04-14:
still not fixed tho.

eDaterKenobi, 2025-04-14:
> still not fixed tho.

yeah cuz they dont view it as a glitch. ):

juanemartinboix, 2025-11-14:
The problem was that I use the step as a way to maintaing the sharpness of the mould flange . Normally you sand the inside of the mould and at very end you "block sand" the step to create the corner. Without the modified 3 walls the model would get too thin after sanding the steps.

Model and strip modifiers 
[image]

Left only 2 perimeters , right 3 perimeters
[image]

juanemartinboix, 2025-11-14:
You have to move the modifier body fully inside (between the thickness of the rest of the walls) in order to not to get extra seams in the outside wall.

[image]

I´m getting extra seams in the flanges were they are OK.