OrcaSlicer/OrcaSlicer #7156: Hybrid Normal and Tree Supports

bjaurelio, 2024-10-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have recently been printing a variety of helmets, and there's one feature I would love to have. Many files have curved bottoms where the supports are low to the build plate. These really need standard supports because the tree trunks are so small they don't work well beginning as tiny dots. However, for features higher up on the models, or where it's not a full dome like an Iron Man helmet, tree supports work much better due to the branch angles and use far less filament. A hybrid approach would be wonderful

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like an option for tree supports that still uses standard supports when the support height is below a threshold. This way, I could have those low curves use standard supports and tree supports everywhere else to get the best of both.

### Describe alternatives you've considered

1. Just use tree supports and hope it starts well.
2. Just use normal supports and deal with the waste or marks on the outside when supports cannot touch the buildplate.

Noisyfox, 2024-10-21:
[image]
Have you tried Tree Hybrid?

bjaurelio, 2024-10-21:
I only have Tree (normal) and Tree (manual).

Noisyfox, 2024-10-22:
Enable advance mode first.

closed 2025-01-28 by github-actions[bot]