OrcaSlicer/OrcaSlicer #7168: Please add a key to Adaptive Layer Height UX

tlhintoq, 2024-10-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Not a problem, just user friendliness

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If you apply `Adaptive layer height` and you see your part ranging from red to green to brown and that tells you... ?  What?
Short of actively hovering over an area in the layers slider there's no real-time indication of the range WHILE making adjustments to the quality and radius sliders.  So there's lots of back and forth mousing go on.
[image]

### Describe alternatives you've considered

If the min/max ticket were handled at the same time (https://github.com/SoftFever/OrcaSlicer/issues/5709) then just showing the min/max on screen would go a long way as well

yw4z, 2024-10-22:
instead using keys maybe a legend for it might be helpful because "auto" feature creates too much variable for thickness values. its showing value of thickness with mouse hover. Current value box is a bit small and easy to miss maybe making it a bit bigger and adding layer info can make it a bit informative

Currently i'm so busy but i will back to continuing where i left when i got some time

tlhintoq, 2024-10-23:
'Key'... 'Legend'... 'Chart'... 'Table'... same thing
[image]

tlhintoq, 2025-01-22:
Would still be really nice to see this added to the tool.

Felix14-v2, 2026-10-07:
> If the min/max ticket were handled at the same time ([#5709](https://github.com/OrcaSlicer/OrcaSlicer/issues/5709)) then just showing the min/max on screen would go a long way as well

The min/max values are taken from the extruder capabilities:

[image]

However I'd like to have a separate temporarily min/max height configuration too, without need to go to the printer settings each time you want to tweak that.