OrcaSlicer/OrcaSlicer #7172: object paining mode not displaying more than 16 filaments

Batalhoti, 2024-10-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 RC

### Operating System (OS)

Windows

### OS Version

windows 11

### Printer

Voron

### How to reproduce

Open Orca Slicer and load a 3D model.

Add 21 filaments.

Access the object painting mode.

Observe that only 16 filaments appear in the painting visualization.

### Actual results

Only 16 filaments appear, while the rest are not displayed

### Expected results

All created filaments should be visible and editable in the object painting mode

### Project file & Debug log uploads

[debug_Sun_Oct_20_17_57_08_31908.log.0.zip](https://github.com/user-attachments/files/17452096/debug_Sun_Oct_20_17_57_08_31908.log.0.zip)
![Captura de tela 2024-10-20 175754](https://github.com/user-attach
[image]
[image]
ments/assets/22b41e2d-329a-4fed-ab40-fbea016c815a)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

When using the object painting mode, I noticed that only 16 filament are displayed in the painting visualization, not the 21 I have defined. This makes it difficult to adjust and finalize the design according to specific needs.

Noisyfox, 2024-10-21:
This is a technical limitation in current code that the max total number of painting colors are limited to 16.

closed 2025-01-28 by github-actions[bot]

DarthLeahcim, 2025-02-12:
I would love to see more than 16 filaments get implemented. 

[image]

Perhaps a good compromise would be the ability to swap which filaments are included in the 16? Maybe this way the technical limitation wouldn't have to get changed. (Ended up finding the exact issue, but I know just enough to be dangerous, not enough to know the correct way fix it or what other issues fixing it would cause)