OrcaSlicer/OrcaSlicer #7173: horizontal vasemode

Student1-29424005, 2024-10-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

nope.  no problem just a suggestion on getting something to work better.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I was thinking, if vasemode is a continual 1 layer thick z climb, why can't we do that on a xy plane?  ie. baselayer prints with concentric pattern but continuously spirals inside.  For the start I think we'd probably have to do a scarf joint though.  

### Describe alternatives you've considered

nope, nothing really to write here.   From what I see there's not much I could find on this idea.

### Additional context

N/A

Student1-29424005, 2024-10-22:
If what I think is correct, this would make an xy layer look much better.  My thoughts were to have vasemode on the xy and not the z.  Imagine a top layer where instead of having a pattern that leave tiny holes between the walls and top layer, the print head would just circle the part like concentric does, but instead of doing a series of continuous loops, you do a single contiguous line (lke vase).  It won't work for all shapes, of course, but if we could very the flow and line width it could probably do quite a good job of making the top layer look better.

closed 2025-01-28 by github-actions[bot]

Felix14-v2, 2025-02-06:
> instead of doing a series of continuous loops, you do a single contiguous line (lke vase). It would eliminate the seamline that a tope concentric pattern would make. It won't work for all shapes, of course, but if we could very the flow and line width it could probably do quite a good job of making the top layer look better.

Your suggestion is identical to the spiral infill option. You can select it in the infill pattern settings and enjoy the quality.